xdspacelab/openvslam #288: Mobile robot navigation

zlrpuck, 2020-04-08:
Hi, 
Is there any tutorial on how to build a mobile robot navigation using Openvslam on ros? 

Thanks

Shashika007, 2020-04-08:
@zlrpuck So far no tutorial. But, you can try to create a grid map using octomap library. For that, you need to output the poincloud of the openvslam. Then, you can use openvslam as your localization method (no amcl) and the grid map to set the goals. Another point is, you need to get the tf transformation of map->odom by openvslam pose data and the odom data. For, obstacle avoidance, you can get the current keyframe pointcloud data then convert into fake lidar data. Then add it to your move_base.

zlin-ir, 2020-04-09:
> @zlrpuck So far no tutorial. But, you can try to create a grid map using octomap library. For that, you need to output the poincloud of the openvslam. Then, you can use openvslam as your [...]

Great! Thanks for your reply.

closed 2020-05-01 by zlrpuck

Zumbalamambo, 2020-06-02:
@Shashika007 is it possible to create gridmap using a monocular video?

mirellameelo, 2020-06-02:
@Shashika007 Tks for sharing! 

The output data from Openvslam is a sparse point cloud, I was wondering if it is an input appropriated to use with this library. Maybe using an RGBD camera to achieve something denser... I'm confused. =/ Could you, please, clarify this point?

@Zumbalamambo check it out #92

Zumbalamambo, 2020-06-05:
@mirellameelo The openvslam gives the 3d position of the camera. But I would like to know where exactly are those positions in the world coordinates for AR application. Are there any other ways to do so?

mirellameelo, 2020-06-08:
@Zumbalamambo I believe what you're asking depends on the origin and 3D axis of your AR system specifically. Once you find that correspondence to OpenCV, you can apply it to the camera's optical center. To know more about the pose estimation process and the relationship between model, camera, and world coordinates, as well as the relationship between the camera space origin and the optical center, please refer to:

- Lepetit, Vincent, and Pascal Fua. Monocular model-based 3D tracking of rigid objects. Now Publishers Inc, 2005.

- Hartley, Richard, and Andrew Zisserman. Multiple view geometry in computer vision. Cambridge university press, 2003.