xdspacelab/openvslam #292: Get Localization frame position

OmarLaz97, 2020-04-21:
Greetings, thank you for this amazing work!!
I was wondering if there is a way to run image localization and extract the position of each frame to use on a 2d map for navigation purposes ?

alexs7, 2020-04-27:
You can have a look at the source code, and print it out.

Shashika007, 2020-05-07:
@OmarLaz97 use this to get the position. Then you can convert it to an odom message. 

auto camera_pose = map_publisher_->get_current_cam_pose().inverse().eval();

OmarLaz97, 2020-05-07:
@Shashika007 thank you for your response, will this allow me to get also the camera orientation (the direction where the camera is looking) ?

Shashika007, 2020-05-07:
@OmarLaz97 yes. this will allow you to output the camera pose matrix (se3) respect to camera to world (get_current_cam_pose()) . For the navigation, you need to get the inverse. This matrix contains 3dof orientation (so3) and 3dof pose. Please check se3 matrix. So, you will understand the structure of the pose that output from this.

OmarLaz97, 2020-05-07:
@Shashika007 okay thank you so much for your reply

closed 2020-05-07 by OmarLaz97

alexs7, 2020-05-07:
@Shashika007 Hello, just wondering why the inverse aswell? The inverse will give you the tx,ty,tz and R in world coordinates right ? The pose itself takes you to camera coordinate system from world in OpenVSLAM right ?

Shashika007, 2020-05-12:
@alexs7 Inverse suppose to give you the transformation matrix of  world (map) frame to camera frame that we need in navigation.  Normally, most of slam system give you the pose from camera to world. So, we need to get the inverse of it.

alexs7, 2020-05-12:
@Shashika007 ok thanks!

chyohoo, 2020-05-17:
> @OmarLaz97 use this to get the position. Then you can convert it to an odom message.
> 
> auto camera_pose = map_publisher_->get_current_cam_pose().inverse().eval();
I want use the position in run_video_slam.cc but I got this
error: ‘map_publisher_’ was not declared in this scope.
How to solve this?

dimaxano, 2020-05-22:
@chyohoo You can get `map_publisher_` object by running
`auto map_publisher_ptr = SLAM.get_map_publisher();`

hetareba, 2020-07-01:
> @chyohoo You can get `map_publisher_` object by running
> `auto map_publisher_ = SLAM.get_map_publisher();`

Which file should you add your code to?
I'm not familiar with C++.
I searched around for files on GitHub but I didn't understand.

dimaxano, 2020-07-02:
Hi, @hetareba! Do you use ROS version of openvslam or not?

hetareba, 2020-07-02:
> Hi, @hetareba! Do you use ROS version of openvslam or not?

No, not yet.  I'm sorry for not studying.
Any hints on the ROS version?

dimaxano, 2020-07-02:
That's ok)
If you are using non-ros version (i.e. `run_video_slam` script), you can add 
this peace of code
```
auto map_publisher_ = SLAM.get_map_publisher();
auto camera_pose = map_publisher_->get_current_cam_pose().inverse().eval();
```
to [this](https://github.com/xdspacelab/openvslam/blob/6de5fec058647297ca3458e4719b42947fd4a582/example/run_video_slam.cc#L66) line