xdspacelab/openvslam #296: What is the Camera.fps value for ?

alexs7, 2020-04-27:
The .yaml file has a param `Camera.fps`. I am trying to run `./example/run_image_slam.cc` with images that where sampled roughly every 400ms. So should the `Camera.fps` be considered at this point or just set to a random value like 10fps / 15fps ?

shinsumicco, 2020-04-28:
`Camera.fps` parameter is currently used for 1) detecting the failure of initialization and 2) feeding a sequence at a constant interval.

1)
https://github.com/xdspacelab/openvslam/blob/4b18d9a17caf24613b1a5cbe6a729bd5cf0e7a5b/src/openvslam/tracking_module.cc#L225-L226

2)
https://github.com/xdspacelab/openvslam/blob/fa7b59ec9d59b34b203e3c64a0ef7916e5628dfc/example/run_video_slam.cc#L78

You can set the parameter as you like if you don't care about the above features.

alexs7, 2020-04-28:
@shinsumicco thanks! I am using `./run_image_slam` to create a model from images. 
In that case what would you advice ?

shinsumicco, 2020-04-28:
For `./run_image_slam`, timestamp of each frame is decided according to a FPS parameter written in the YAML configuration.
https://github.com/xdspacelab/openvslam/blob/5a0b1a5f52b4d29b699624052c9d5dc4417d9882/example/util/image_util.cc#L28-L30

If frames are fed to the SLAM system so frequently that you failure the mapping, please decrease the FPS value to assure the feeding interval.

alexs7, 2020-04-28:
@shinsumicco ok thanks!

closed 2020-04-28 by alexs7