xdspacelab/openvslam #301: double free or corruption

shrine1112, 2020-05-07:
[2020-05-06 19:04:40.705] [I] loading ORB vocabulary: ./3rd/DBoW2/orb_vocab/orb_vocab.dbow2
[2020-05-06 19:04:41.164] [I] startup SLAM system
[2020-05-06 19:04:41.165] [I] start mapping module
[2020-05-06 19:04:41.165] [I] start global optimization module
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.X11 Error: GLXBadFBConfig
[2020-05-06 19:04:44.199] [I] initialization succeeded with H
*** Error in `./build/run_euroc_slam': double free or corruption (out): 0x00007f118831f680 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f11af3647e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f11af36d37a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f11af37153c]
/home/shrine/vslam_ws/VO/openvslam/3rd/usr/local/lib/libg2o_core.so(_ZN3g2o17JacobianWorkspaceD2Ev+0x24)[0x7f11ab148744]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN3g2o11BlockSolverINS_17BlockSolverTraitsILi6ELi3EEEE11buildSystemEv+0x20f)[0x7f11b1eb5abf]
/home/shrine/vslam_ws/VO/openvslam/3rd/usr/local/lib/libg2o_core.so(_ZN3g2o30OptimizationAlgorithmLevenberg5solveEib+0x3e9)[0x7f11ab143059]
/home/shrine/vslam_ws/VO/openvslam/3rd/usr/local/lib/libg2o_core.so(_ZN3g2o15SparseOptimizer8optimizeEib+0xad)[0x7f11ab13aa7d]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZNK9openvslam8optimize22global_bundle_adjuster8optimizeEjPb+0x775)[0x7f11b1ef4045]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN9openvslam6module11initializer24create_map_for_monocularERNS_4data5frameE+0x589)[0x7f11b1ea0949]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN9openvslam6module11initializer10initializeERNS_4data5frameE+0x53)[0x7f11b1ea1603]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN9openvslam15tracking_module10initializeEv+0x1c)[0x7f11b1e0f0bc]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN9openvslam15tracking_module5trackEv+0x775)[0x7f11b1e10f75]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN9openvslam15tracking_module21track_monocular_imageERKN2cv3MatEdS4_+0xe2)[0x7f11b1e113b2]
/home/shrine/vslam_ws/VO/openvslam/build/lib/libopenvslam.so(_ZN9openvslam6system20feed_monocular_frameERKN2cv3MatEdS4_+0x4c)[0x7f11b1e03a8c]
./build/run_euroc_slam[0x41a0b1]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xd0840)[0x7f11af99f840]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f11b1b4f6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f11af3f441d]
======= Memory map: ========
00400000-0044b000 r-xp 00000000 08:01 3826960                            /home/shrine/vslam_ws/VO/openvslam/build/run_euroc_slam
0064b000-0064c000 r--p 0004b000 08:01 3826960                            /home/shrine/vslam_ws/VO/openvslam/build/run_euroc_slam
0064c000-0064d000 rw-p 0004c000 08:01 3826960                            /home/shrine/vslam_ws/VO/openvslam/build/run_euroc_slam
0064d000-0064e000 rw-p 00000000 00:00 0 
01c53000-12e09000 rw-p 00000000 00:00 0                                  [heap]
7f1144000000-7f11441e3000 rw-p 00000000 00:00 0 
7f11441e3000-7f1148000000 ---p 00000000 00:00 0 
7f114c000000-7f114c022000 rw-p 00000000 00:00 0 
7f114c022000-7f1150000000 ---p 00000000 00:00 0 
7f1150000000-7f11501f7000 rw-p 00000000 00:00 0 
7f11501f7000-7f1154000000 ---p 00000000 00:00 0 
7f1154000000-7f11541f3000 rw-p 00000000 00:00 0 
7f11541f3000-7f1158000000 ---p 00000000 00:00 0 
7f115c000000-7f115c022000 rw-p 00000000 00:00 0 
7f115c022000-7f1160000000 ---p 00000000 00:00 0 
7f1160c97000-7f1160e1b000 rw-s 00000000 00:05 41025551                   /SYSV00000000 (deleted)
7f1160e1b000-7f1160f1b000 rw-s 10d2ba000 00:06 331                       /dev/dri/card0
7f1160f1b000-7f1160f1c000 ---p 00000000 00:00 0 
7f1160f1c000-7f116171c000 rwxp 00000000 00:00 0 
7f116171c000-7f116171d000 ---p 00000000 00:00 0 
7f116171d000-7f1161f1d000 rwxp 00000000 00:00 0 
7f1161f1d000-7f1161f53000 r-xp 00000000 08:01 4064707                    /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so
7f1161f53000-7f1162153000 ---p 00036000 08:01 4064707                    /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so
7f1162153000-7f1162158000 r--p 00036000 08:01 4064707                    /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so
7f1162158000-7f1162159000 rw-p 0003b000 08:01 4064707                    /usr/lib/x86_64-linux-gnu/gvfs/libgvfscommon.so
7f1162159000-7f1162189000 r-xp 00000000 08:01 4064708                    /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
7f1162189000-7f1162389000 ---p 00030000 08:01 4064708                    /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
7f1162389000-7f116238a000 r--p 00030000 08:01 4064708                    /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
7f116238a000-7f116238c000 rw-p 00031000 08:01 4064708                    /usr/lib/x86_64-linux-gnu/gio/modules/libgvfsdbus.so
7f116238c000-7f11623f2000 r-xp 00000000 08:01 3801804                    /usr/lib/x86_64-linux-gnu/libibus-1.0.so.5.0.511
7f11623f2000-7f11625f1000 ---p 00066000 08:01 3801804                    /usr/lib/x86_64-linux-gnu/libibus-1.0.so.5.0.511
7f11625f1000-7f11625f3000 r--p 00065000 08:01 3801804                    /usr/lib/x86_64-linux-gnu/libibus-1.0.so.5.0.511
7f11625f3000-7f11625f4000 rw-p 00067000 08:01 3801804                    /usr/lib/x86_64-linux-gnu/libibus-1.0.so.5.0.511
7f11625f4000-7f11625f5000 rw-p 00000000 00:00 0 
7f11625f5000-7f11625fb000 r-xp 00000000 08:01 4065134                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so
7f11625fb000-7f11627fb000 ---p 00006000 08:01 4065134                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so
7f11627fb000-7f11627fc000 r--p 00006000 08:01 4065134                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so
7f11627fc000-7f11627fd000 rw-p 00007000 08:01 4065134                    /usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so
7f11627fd000-7f11627fe000 ---p 00000000 00:00 0 
7f11627fe000-7f1162ffe000 rwxp 00000000 00:00 0 
7f1162ffe000-7f1162fff000 ---p 00000000 00:00 0 
7f1162fff000-7f11637ff000 rwxp 00000000 00:00 0 
7f11637ff000-7f1163800000 ---p 00000000 00:00 0 
[... 462 lines cut ...]
PC: @     0x7f11af322428 gsignal
*** SIGABRT (@0x3e800005514) received by PID 21780 (TID 0x7f118ed01700) from PID 21780; stack trace: ***
    @     0x7f11b1b59390 (unknown)
    @     0x7f11af322428 gsignal
    @     0x7f11af32402a abort
    @     0x7f11af3647ea (unknown)
    @     0x7f11af36d37a (unknown)
    @     0x7f11af37153c cfree
    @     0x7f11ab148744 g2o::JacobianWorkspace::~JacobianWorkspace()
    @     0x7f11b1eb5abf g2o::BlockSolver<>::buildSystem()
    @     0x7f11ab143059 g2o::OptimizationAlgorithmLevenberg::solve()
    @     0x7f11ab13aa7d g2o::SparseOptimizer::optimize()
    @     0x7f11b1ef4045 openvslam::optimize::global_bundle_adjuster::optimize()
    @     0x7f11b1ea0949 openvslam::module::initializer::create_map_for_monocular()
    @     0x7f11b1ea1603 openvslam::module::initializer::initialize()
    @     0x7f11b1e0f0bc openvslam::tracking_module::initialize()
    @     0x7f11b1e10f75 openvslam::tracking_module::track()
    @     0x7f11b1e113b2 openvslam::tracking_module::track_monocular_image()
    @     0x7f11b1e03a8c openvslam::system::feed_monocular_frame()
    @           0x41a0b1 _ZZ13mono_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_jbbbSC_bENKUlvE_clEv
    @     0x7f11af99f840 (unknown)
    @     0x7f11b1b4f6ba start_thread
    @     0x7f11af3f441d clone
    @                0x0 (unknown)
Aborted (core dumped)

shinsumicco, 2020-05-07:
Did you try this?
https://openvslam.readthedocs.io/en/master/trouble_shooting.html

shrine1112, 2020-05-07:
it works for me, thanks!

shinsumicco, 2020-05-07:
I'm glad to hear that!

closed 2020-05-07 by shinsumicco