xdspacelab/openvslam #303: Can't launch slam without sudo. Can't launch ros node

d-desiatkin, 2020-05-08:
### Without sudo:

> robotics@jetson-tx2:~/porter_ws/src/openvslam/build$` ./run_euroc_slam     -v ../ros/config/orb_vocab.dbow2     -d ~/Downloads/mav0/     -c ../example/euroc/EuRoC_mono.yaml --map-db /home/robotics/map.msg --debug
> 
> [2020-05-08 16:43:29.448] [D] CONSTRUCT: config
> [2020-05-08 16:43:29.448] [I] config file loaded: ../example/euroc/EuRoC_mono.yaml
> [2020-05-08 16:43:29.448] [D] load camera model type
> [2020-05-08 16:43:29.448] [D] load camera model parameters
> [2020-05-08 16:43:29.449] [D] CONSTRUCT: camera::base
> [2020-05-08 16:43:29.449] [D] CONSTRUCT: camera::perspective
> [2020-05-08 16:43:29.449] [D] compute image bounds
> [2020-05-08 16:43:29.449] [D] load ORB parameters
> [2020-05-08 16:43:29.450] [D] load tracking parameters
> [2020-05-08 16:43:29.450] [D] load depth threshold
> [2020-05-08 16:43:29.450] [D] load depthmap factor
> [2020-05-08 16:43:29.486] [D] CONSTRUCT: system
>   ___               __   _____ _      _   __  __ 
>  / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
> | (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
>  \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
>       |_|                                        
> 
> Copyright (C) 2019,
> National Institute of Advanced Industrial Science and Technology (AIST)
> All rights reserved.
> 
> This is free software,
> and you are welcome to redistribute it under certain conditions.
> See the LICENSE file.
> 
> Camera Configuration:
> - name: EuRoC monocular
> - setup: Monocular
> - fps: 20
> - cols: 752
> - rows: 480
> - color: Gray
> - model: Perspective
>   - fx: 458.654
>   - fy: 457.296
>   - cx: 367.215
>   - cy: 248.375
>   - k1: -0.283408
>   - k2: 0.0739591
>   - p1: 0.00019359
>   - p2: 1.76187e-05
>   - k3: 0
>   - min x: -135.812
>   - max x: 895.507
>   - min y: -92.8751
>   - max y: 565.556
> ORB Configuration:
> - number of keypoints: 1000
> - scale factor: 1.2
> - number of levels: 8
> - initial fast threshold: 20
> - minimum fast threshold: 7
> 
> [2020-05-08 16:43:29.487] [I] loading ORB vocabulary: ../ros/config/orb_vocab.dbow2
> [2020-05-08 16:43:31.338] [D] CONSTRUCT: data::camera_database
> [2020-05-08 16:43:31.338] [D] CONSTRUCT: data::map_database
> [2020-05-08 16:43:31.338] [D] CONSTRUCT: data::bow_database
> [2020-05-08 16:43:31.341] [D] CONSTRUCT: publish::frame_publisher
> [2020-05-08 16:43:31.342] [D] CONSTRUCT: publish::map_publisher
> [2020-05-08 16:43:31.342] [D] CONSTRUCT: module::initializer
> [2020-05-08 16:43:31.342] [D] CONSTRUCT: module::relocalizer
> [2020-05-08 16:43:31.342] [D] CONSTRUCT: tracking_module
> [2020-05-08 16:43:31.342] [D] CONSTRUCT: mapping_module
> [2020-05-08 16:43:31.342] [D] CONSTRUCT: global_optimization_module
> [2020-05-08 16:43:31.342] [I] startup SLAM system
> [2020-05-08 16:43:31.343] [I] start mapping module
> [2020-05-08 16:43:31.343] [I] start global optimization module
> [2020-05-08 16:43:31.446] [D] CONSTRUCT: initialize::perspective
> Gtk-Message: 16:43:31.752: Failed to load module "canberra-gtk-module"
> [2020-05-08 16:43:41.897] [I] initialization succeeded with F
> [2020-05-08 16:43:41.897] [D] DESTRUCT: initialize::perspective

**And here all just freeze.**

### With sudo:

> robotics@jetson-tx2:~/porter_ws/src/openvslam/build$ sudo ./run_euroc_slam     -v ../ros/config/orb_vocab.dbow2     -d ~/Downloads/mav0/     -c ../example/euroc/EuRoC_mono.yaml --map-db /home/robotics/map.msg --debug
> [2020-05-08 16:47:44.264] [D] CONSTRUCT: config
> [2020-05-08 16:47:44.265] [I] config file loaded: ../example/euroc/EuRoC_mono.yaml
> [2020-05-08 16:47:44.265] [D] load camera model type
> [2020-05-08 16:47:44.265] [D] load camera model parameters
> [2020-05-08 16:47:44.265] [D] CONSTRUCT: camera::base
> [2020-05-08 16:47:44.265] [D] CONSTRUCT: camera::perspective
> [2020-05-08 16:47:44.265] [D] compute image bounds
> [2020-05-08 16:47:44.266] [D] load ORB parameters
> [2020-05-08 16:47:44.266] [D] load tracking parameters
> [2020-05-08 16:47:44.266] [D] load depth threshold
> [2020-05-08 16:47:44.266] [D] load depthmap factor
> [2020-05-08 16:47:44.309] [D] CONSTRUCT: system
>   ___               __   _____ _      _   __  __ 
>  / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
> | (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
>  \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
>       |_|                                        
> 
> Copyright (C) 2019,
> National Institute of Advanced Industrial Science and Technology (AIST)
> All rights reserved.
> 
> This is free software,
> and you are welcome to redistribute it under certain conditions.
> See the LICENSE file.
> 
> Camera Configuration:
> - name: EuRoC monocular
> - setup: Monocular
> - fps: 20
> - cols: 752
> - rows: 480
> - color: Gray
> - model: Perspective
>   - fx: 458.654
>   - fy: 457.296
>   - cx: 367.215
>   - cy: 248.375
>   - k1: -0.283408
>   - k2: 0.0739591
>   - p1: 0.00019359
>   - p2: 1.76187e-05
>   - k3: 0
>   - min x: -135.812
>   - max x: 895.507
>   - min y: -92.8751
>   - max y: 565.556
> ORB Configuration:
> - number of keypoints: 1000
> - scale factor: 1.2
> - number of levels: 8
> - initial fast threshold: 20
> - minimum fast threshold: 7
> 
> [2020-05-08 16:47:44.311] [I] loading ORB vocabulary: ../ros/config/orb_vocab.dbow2
> [2020-05-08 16:47:46.208] [D] CONSTRUCT: data::camera_database
> [2020-05-08 16:47:46.208] [D] CONSTRUCT: data::map_database
> [2020-05-08 16:47:46.208] [D] CONSTRUCT: data::bow_database
> [2020-05-08 16:47:46.212] [D] CONSTRUCT: publish::frame_publisher
> [2020-05-08 16:47:46.213] [D] CONSTRUCT: publish::map_publisher
> [2020-05-08 16:47:46.213] [D] CONSTRUCT: module::initializer
> [2020-05-08 16:47:46.213] [D] CONSTRUCT: module::relocalizer
> [2020-05-08 16:47:46.213] [D] CONSTRUCT: tracking_module
> [2020-05-08 16:47:46.213] [D] CONSTRUCT: mapping_module
> [2020-05-08 16:47:46.213] [D] CONSTRUCT: global_optimization_module
> [2020-05-08 16:47:46.213] [I] startup SLAM system
> [2020-05-08 16:47:46.214] [I] start mapping module
> [2020-05-08 16:47:46.214] [I] start global optimization module
> error: XDG_RUNTIME_DIR not set in the environment.
> [2020-05-08 16:47:46.338] [D] CONSTRUCT: initialize::perspective
> Gtk-Message: 16:47:46.647: Failed to load module "canberra-gtk-module"
> [2020-05-08 16:47:56.096] [I] initialization succeeded with F
> [2020-05-08 16:47:56.096] [D] DESTRUCT: initialize::perspective
> [2020-05-08 16:47:56.193] [I] new map created with 94 points: frame 0 - frame 111

**Here everything is fine.**

Basically I have the problem with launching ros node. I found this behavior and can't make it work.

g2o was compiled with and without BUILD_WITH_MARCH_NATIVE

d-desiatkin, 2020-05-08:
### UPDATE

I recompiled openvslam with address sanitizers. Here is the output:

> robotics@jetson-tx2:~/porter_ws/src/openvslam/build$ sudo ./run_euroc_slam     -v ../ros/config/orb_vocab.dbow2     -d ~/Downloads/mav0/     -c ../example/euroc/EuRoC_mono.yaml --map-db /home/robotics/map.msg --debug
> [2020-05-08 19:27:59.388] [D] CONSTRUCT: config
> [2020-05-08 19:27:59.388] [I] config file loaded: ../example/euroc/EuRoC_mono.yaml
> [2020-05-08 19:27:59.389] [D] load camera model type
> [2020-05-08 19:27:59.389] [D] load camera model parameters
> [2020-05-08 19:27:59.389] [D] CONSTRUCT: camera::base
> [2020-05-08 19:27:59.389] [D] CONSTRUCT: camera::perspective
> [2020-05-08 19:27:59.389] [D] compute image bounds
> [2020-05-08 19:27:59.389] [D] load ORB parameters
> [2020-05-08 19:27:59.390] [D] load tracking parameters
> [2020-05-08 19:27:59.390] [D] load depth threshold
> [2020-05-08 19:27:59.390] [D] load depthmap factor
> [2020-05-08 19:27:59.475] [D] CONSTRUCT: system
>   ___               __   _____ _      _   __  __ 
>  / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
> | (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
>  \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
>       |_|                                        
> 
> Copyright (C) 2019,
> National Institute of Advanced Industrial Science and Technology (AIST)
> All rights reserved.
> 
> This is free software,
> and you are welcome to redistribute it under certain conditions.
> See the LICENSE file.
> 
> Camera Configuration:
> - name: EuRoC monocular
> - setup: Monocular
> - fps: 20
> - cols: 752
> - rows: 480
> - color: Gray
> - model: Perspective
>   - fx: 458.654
>   - fy: 457.296
>   - cx: 367.215
>   - cy: 248.375
>   - k1: -0.283408
>   - k2: 0.0739591
>   - p1: 0.00019359
>   - p2: 1.76187e-05
>   - k3: 0
>   - min x: -135.812
>   - max x: 895.507
>   - min y: -92.8751
>   - max y: 565.556
> ORB Configuration:
> - number of keypoints: 1000
> - scale factor: 1.2
> - number of levels: 8
> - initial fast threshold: 20
> - minimum fast threshold: 7
> 
> [2020-05-08 19:27:59.476] [I] loading ORB vocabulary: ../ros/config/orb_vocab.dbow2
> [2020-05-08 19:28:02.503] [D] CONSTRUCT: data::camera_database
> [2020-05-08 19:28:02.503] [D] CONSTRUCT: data::map_database
> [2020-05-08 19:28:02.503] [D] CONSTRUCT: data::bow_database
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: publish::frame_publisher
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: publish::map_publisher
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: module::initializer
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: module::relocalizer
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: tracking_module
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: mapping_module
> [2020-05-08 19:28:02.505] [D] CONSTRUCT: global_optimization_module
[... 31 lines cut ...]
>     #3 0x7f9716c2cf in g2o::OptimizationAlgorithmLevenberg::solve(int, bool) (/usr/local/lib/libg2o_core.so+0x462cf)
>     #4 0x7f97164d5f in g2o::SparseOptimizer::optimize(int, bool) (/usr/local/lib/libg2o_core.so+0x3ed5f)
>     #5 0x7f98de3483 in openvslam::optimize::global_bundle_adjuster::optimize(unsigned int, bool*) const (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x507483)
>     #6 0x7f98c953e7 in openvslam::module::initializer::create_map_for_monocular(openvslam::data::frame&) (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x3b93e7)
>     #7 0x7f98ca028b in openvslam::module::initializer::initialize(openvslam::data::frame&) (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x3c428b)
>     #8 0x7f98a5a7af in openvslam::tracking_module::initialize() (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x17e7af)
>     #9 0x7f98a65cb3 in openvslam::tracking_module::track() (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x189cb3)
>     #10 0x7f98a67697 in openvslam::tracking_module::track_monocular_image(cv::Mat const&, double, cv::Mat const&) (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x18b697)
>     #11 0x7f98a360ef in openvslam::system::feed_monocular_frame(cv::Mat const&, double, cv::Mat const&) (/home/robotics/porter_ws/src/openvslam/build/lib/libopenvslam.so+0x15a0ef)
>     #12 0x55562fcb43 in mono_tracking(std::shared_ptr<openvslam::config> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, bool, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda()#1}::operator()() const (/home/robotics/porter_ws/src/openvslam/build/run_euroc_slam+0x27b43)
>     #13 0x55562fd817 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<mono_tracking(std::shared_ptr<openvslam::config> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, bool, bool, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)::{lambda()#1}> > >::_M_run() (/home/robotics/porter_ws/src/openvslam/build/run_euroc_slam+0x28817)
>     #14 0x7f97c2550b  (/usr/lib/aarch64-linux-gnu/libstdc++.so.6+0xcc50b)
>     #15 0x7f988b7087 in start_thread (/lib/aarch64-linux-gnu/libpthread.so.0+0x7087)
>     #16 0x7f97aab4eb  (/lib/aarch64-linux-gnu/libc.so.6+0xd04eb)
> 
[... cut ...] :allocator<char> > const&, bool) (/home/robotics/porter_ws/src/openvslam/build/run_euroc_slam+0x293af)
>     #3 0x55562f6dbb in main (/home/robotics/porter_ws/src/openvslam/build/run_euroc_slam+0x21dbb)
>     #4 0x7f979fb6df in __libc_start_main (/lib/aarch64-linux-gnu/libc.so.6+0x206df)
>     #5 0x55562f7bff  (/home/robotics/porter_ws/src/openvslam/build/run_euroc_slam+0x22bff)
> 
> SUMMARY: AddressSanitizer: bad-free (/usr/lib/aarch64-linux-gnu/libasan.so.4+0xcf77f) in free
> ==19059==ABORTING

d-desiatkin, 2020-05-09:
### UPDATE

I found out a problem. It seems that build g2o with OpenMP leads to an error. But I don't understand the reason. I have successfully built your project on my personal notebook. It seems that this issue connected with architecture.

d-desiatkin, 2020-05-09:
### UPDATE
Yeah, ros node is also working as it should now.

closed 2020-05-09 by d-desiatkin

sudo-panda, 2020-07-11:
@d-desiatkin I tried building g2o without OpenMP but it didn't work. Are you sure you didn't do anything else?