xdspacelab/openvslam #304: Unable to run the example

Zumbalamambo, 2020-05-08:
It throws the following error while I'm trying to install,

```
> terminate called after throwing an instance of 'std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >'
> *** Aborted at 1588965076 (unix time) try "date -d @1588965076" if you are using GNU date ***
> PC: @     0x7f159dbdee97 gsignal
> *** SIGABRT (@0x3e800002a27) received by PID 10791 (TID 0x7f15a0c71000) from PID 10791; stack trace: ***
>     @     0x7f159fb92890 (unknown)
>     @     0x7f159dbdee97 gsignal
>     @     0x7f159dbe0801 abort
>     @     0x7f159e25a227 (unknown)
>     @     0x7f159e2655d6 (unknown)
>     @     0x7f159e265641 std::terminate()
>     @     0x7f159e2658d5 __cxa_throw
>     @     0x7f15a0382cf8 DBoW2::TemplatedVocabulary<>::loadFromBinaryFile()
>     @     0x7f15a03801b4 openvslam::system::system()
>     @     0x7f15a101aae6 mono_tracking()
>     @     0x7f15a10195d0 main
>     @     0x7f159dbc1b97 __libc_start_main
>     @     0x7f15a1019afa _start
> Aborted (core dumped)
```

ymd-stella, 2020-05-09:
Is orb_vocab.dbow2 obtained by the procedure described in [simple_tutorial.rst](https://github.com/xdspacelab/openvslam/blob/master/docs/simple_tutorial.rst) located in the path specified after `-v` option?

Zumbalamambo, 2020-05-09:
@ymd-stella, of course, it is there... :(

ymd-stella, 2020-05-10:
@Zumbalamambo Please show me the command line to reproduce it.

Zumbalamambo, 2020-05-10:
I have used this

`./run_video_localization -v ./orb_vocab/orb_vocab.dbow2 -m ./a_loc/video.mp4 -c ./a_loc/config.yaml --frame-skip 3 --no-sleep --map-db map.msg
`

ymd-stella, 2020-05-10:
Why are you using run_video_localization? (Map required)
Has run_video_slam already been succeeded?

ymd-stella, 2020-05-10:
Try executing `ls -s ./orb_vocab/orb_vocab.dbow2` to make sure the file exists and is correct.
(In my environment, the file size was 43332 bytes.)

closed 2020-05-12 by Zumbalamambo