xdspacelab/openvslam #305: video_slam error running error

liuzheng22, 2020-05-09:
I can successfully run video_slam on demo videos,  but I'm trying to run by on my own video, got the error below. 

`[2020-05-08 09:33:43.597] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
[2020-05-08 09:33:43.945] [I] startup SLAM system
[2020-05-08 09:33:43.945] [I] start mapping module
[2020-05-08 09:33:43.945] [I] start global optimization module
[2020-05-08 09:33:43] [connect] Successful connection
[2020-05-08 09:33:43] [connect] WebSocket Connection 172.17.0.2:3000 v-2 "WebSocket++/0.8.1" /socket.io/?EIO=4&transport=websocket&t=1588930423 101
[2020-05-08 09:33:43.948] [I] connected to server
[h264 @ 0x5648d275c660] Invalid NAL unit size (6204 > 3750).
[h264 @ 0x5648d275c660] Error splitting the input into NAL units.
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5648d27140a0] stream 1, offset 0x3fbea9: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5648d27140a0] stream 1, offset 0x3fc06a: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5648d27140a0] stream 1, offset 0x3fc216: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5648d27140a0] stream 1, offset 0x3fc469: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x5648d27140a0] stream 1, offset 0x3fc6b2: partial file`

Shashika007, 2020-05-12:
@liuzheng22 what is your video file format ?

liuzheng22, 2020-05-13:
@Shashika007 .mp4 file - same to the sample videos

Shivani1796, 2020-05-28:
Hi.. @Shashika007 @liuzheng22  Can the video file run in .bag format ??