xdspacelab/openvslam #307: Retrieve Camera Postion

Zumbalamambo, 2020-05-10:
How do I retrieve the camera position in  `run_video_localization` ?

Shashika007, 2020-05-12:
@Zumbalamambo use this to get the position. Then you can convert it to an odom message.

auto camera_pose = map_publisher_->get_current_cam_pose().inverse().eval();

Zumbalamambo, 2020-05-14:
It gives me the following matrix,

```

  0.908122 0.00536217  -0.418671  -0.155941
-0.0407793  -0.994032  -0.101184  0.0249973
  0.416715   -0.10896   0.902483 -0.0830229
        -0         -0         -0         -1

```

How do I interpret them? Where is the rotation matrix, xyz param, orientation of the camera?

dimaxano, 2020-05-15:
1. Yes, you are right
2. This is position of your camera on your map. I think, it is possible to say that these are world coordinates

chyohoo, 2020-05-16:
`rot_cw_ = cam_pose_cw_.block<3, 3>(0, 0);`
`rot_wc_ = rot_cw_.transpose();`
`trans_cw_ = cam_pose_cw_.block<3, 1>(0, 3);`
`cam_center_ = -rot_cw_.transpose() * trans_cw_; `

Zumbalamambo, 2020-05-28:
@dimaxano how do I retrieve the absolute position and the origin?

Zumbalamambo, 2020-05-28:
@dimaxano @chyohoo is cam_center the origin?

AndrewSerbin, 2020-06-12:
Hi guys, @dimaxano @chyohoo . Do you know how to get camera position, rotation matrix, xyz param, orientation of the camera from the saved map (map.msg)? There are rot_cw and trans cw arrays but how to use them? 
`"trans_cw": [
        0.001168138611814195,
        0.001744156806106644,
        -0.04382340588891053
      ],
      "rot_cw": [
        -0.013637094956248765,
        0.00030411659091322845,
        -0.001853422243378439,
        0.9999052465010078
      ],`

dimaxano, 2020-06-18:
Hi, @AndrewSerbin! Use code @chyohoo posted above: you alredy have `rot_cw_` and `trans_cw_` variables. After running the code, `rot_wc_` will contain your camera orientation and `cam_center_` will contain your camera position

Zumbalamambo, 2020-06-18:
@dimaxano how can I use this camera pose information, when the map has got scale drift?..let's say for indoor navigation