xdspacelab/openvslam #309: interpreting output messagePack file

angussmitchell, 2020-05-11:
Hi, I'm trying to interpret information from the generated map for post processing. Ideally I'd like access to the camera trajectory, and sparse cloud points. 

So far I've unpacked the binarized data using a messagePack python library. Now, exploring the data i get, it is not clear where the camera rotations and translations are stored, if they are stored at all. 

Viewing in the python debugger I get the following output:
[image]

by plotting the 'pos_w' values inside the landmarks field I get plot of the sparse cloud (what I wanted), but I am still unable to find any camera rotations or translations. You can see this below:

[image]

Some documentation on how to extract this information would be greatly appreciated.

angussmitchell, 2020-05-11:
the keyframes item contains the camera pose and rotation. This is in TUM format.

closed 2020-05-11 by angussmitchell

Zumbalamambo, 2020-05-13:
@angussmitchell how did you recover the pose and rotation matrix from TUM Format?

angussmitchell, 2020-05-13:
More info on the TUM format can be found here:
https://github.com/MichaelGrupp/evo/wiki/Formats