xdspacelab/openvslam #310: SLAM example bug

pfedom, 2020-05-12:
Hello guys,

I successfully installed OpenVSLAM on Ubuntu 18.04 and i wanted to try out some of the examples. 
I had to turn DBUILD_WITH_MARCH_NATIVE=OFF because OpenVslam terminated every time i tried to start it. 
After building g2o and openvslam, it started, but I get the mapping feature like shown in the example only once like every 10 times. And even when it works its not from start I always have to press reset 2-3 times and toggle mapping on/off. 
 
The other times i see the video in the Frame Viewer with yellow squares in it and on the Map Viewer i can see the blue/grey "triangle" that just keeps moving out of the screen after i start or press reset. I dont get the line made of green triangles or the point clouds like shown in the examples. 

If i get it (however) working it is pretty similar to the example so i think than everything is working fine. 

I am using the fisheye example "nu_eng_corridor_1". 
I also installed the Ros Version for eloquent according to #247 but it doesnt matter if i start it the Ros version or as shown in the tutorial. 

I think the localization is working because the triangle moves but the Mapping function doesnt? 
Is this a known bug? 
Does anyone know on what this bug depends and how i can solve it? 

Thanks in advance!

Shashika007, 2020-05-12:
@pfedom Please explain the steps you run the repo. If it does not works on your laptop maybe you can give a try on openvslam docker.  If you can, please share a screenshot of your bug.

pfedom, 2020-05-12:
@Shashika007 I use this to run the Slam:
[image]
I downloaded the Dataset from the "Simple Tutorial" chapter in the documentation.  

Map Viewer and Frame Viewer open, and the blue/gray waypoint/triangle moves out of the window.
For this screenshot i reset and paused. Before resetting the terminal was in the line "new map created with 244 points: frame 0 - frame 7"

[image]

I will try the docker but it would be nice if it works on the Laptop too.

pfedom, 2020-05-13:
Update: 
openvslam docker runs fluid and the explained bug doesnt appear. 
Is there a way to get the Ros implementation running with the docker application?

Shashika007, 2020-05-14:
@pfedom yes. you have to change the docker file. example - 
```
# FROM ubuntu:18.04 
FROM ros:kinetic
```
then create a catkin workspace and build cv_bridge from source and build opevslam ros as mentioned in tutorial. 

Then you can commit your docker file and save all the changes you have done. 

I recommend you to learn some docker first. there are some good tutorials  for docker ros. please follow those first then you will understand how to change it according to your needs.

pfedom, 2020-05-14:
Thank you, i will try this. 
One last question to the main problem. Is it possible that i did something wrong in the installation? Because its running completely normal in the docker?