xdspacelab/openvslam #311: run_video_slam howto

ai04yc01, 2020-05-13:
First of all, thanks for creating this tool.

I followed the instructions on running run_video_slam. I used a Ricoh Theta S to record a movie, convert it to mono and saved in MP4 format. Copied a yaml from a lab example to try tracking and mapping. But the operation seems stuck at "connected to server".
__________
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2020-05-13 13:30:07.705] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
[2020-05-13 13:30:08.072] [D] CONSTRUCT: data::camera_database
[2020-05-13 13:30:08.072] [D] CONSTRUCT: data::map_database
[2020-05-13 13:30:08.072] [D] CONSTRUCT: data::bow_database
[2020-05-13 13:30:08.072] [D] CONSTRUCT: publish::frame_publisher
[2020-05-13 13:30:08.072] [D] CONSTRUCT: publish::map_publisher
[2020-05-13 13:30:08.072] [D] CONSTRUCT: module::initializer
[2020-05-13 13:30:08.072] [D] CONSTRUCT: module::relocalizer
[2020-05-13 13:30:08.072] [D] CONSTRUCT: tracking_module
[2020-05-13 13:30:08.072] [D] CONSTRUCT: mapping_module
[2020-05-13 13:30:08.072] [D] CONSTRUCT: global_optimization_module
[2020-05-13 13:30:08.072] [I] startup SLAM system
[2020-05-13 13:30:08.073] [I] start mapping module
[2020-05-13 13:30:08.073] [I] start global optimization module
[2020-05-13 13:30:08] [connect] Successful connection
[2020-05-13 13:30:08] [connect] WebSocket Connection 127.0.0.1:3000 v-2 "WebSocket++/0.8.1" /socket.io/?EIO=4&transport=websocket&t=1589347808 101
[2020-05-13 13:30:08.074] [I] connected to server
____________

I'm running it in a VM on MacOS and assigned 4 cores to it. I noticed that the script is actually running. So I guess it is my Mac is not fast enough. What is the minimum config? On the other hand, did I miss any steps or is it normal that the process stuck there?

Shashika007, 2020-05-14:
@ai04yc01 did you try access 127.0.0.1:3000/ from your browser and check the results ?

ai04yc01, 2020-05-14:
Yes I did and the result of the sample aist labs are successful.

ai04yc01, 2020-05-14:
I finally figured out: I have to use the movie in MOV format instead of MP4.

closed 2020-05-14 by ai04yc01