xdspacelab/openvslam #314: Segmentation fault on YAML parse at startup on raspberry pi

keredson, 2020-05-14:
fresh build, first example (on a raspberry pi 3):

```
$ ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg
[2020-05-14 02:38:26.504] [I] config file loaded: ./aist_living_lab_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2020-05-14 02:38:26.509] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
*** Aborted at 1589420308 (unix time) try "date -d @1589420308" if you are using GNU date ***
PC: @ 0x76d0e058 YAML::detail::node::get<>()
*** SIGSEGV (@0x30) received by PID 2051 (TID 0x6d7522a0) from PID 48; stack trace: ***
    @ 0x75082130 (unknown)
    @ 0x76d0e058 YAML::detail::node::get<>()
    @ 0x76d7196c openvslam::module::initializer::initializer()
    @ 0x76cf9924 openvslam::tracking_module::tracking_module()
    @ 0x76cede0c openvslam::system::system()
    @    0x24158 mono_tracking()
    @    0x22bb0 main
    @ 0x7506c718 __libc_start_main
Segmentation fault
```

Shashika007, 2020-05-14:
@keredson 
Did you build yaml-cpp package correctly ? if you do it correctly, try change your CMakeList find_package(yaml-cpp ...) line into following

# yaml-cpp
find_package(YamlCpp REQUIRED)
include_directories(${YAMLCPP_INCLUDE_DIRS})

then build it and run it again.

shinsumicco, 2020-05-19:
Thank you so much for providing a solution, @Shashika007
Solved? @keredson