xdspacelab/openvslam #316: Run on remote server

minhpvo, 2020-05-15:
Hello, 

Thank for the great work.  I wonder if there is a way to run the system without the display (on remote server).

Thank you!

dimaxano, 2020-05-15:
Hi!
I also was wondering in that and the only way I found is to build OpenVSLAM with `-DUSE_PANGOLIN_VIEWER=OFF -DUSE_SOCKET_PUBLISHER=OFF` flags.
But I didn't try that out yet

Zumbalamambo, 2020-05-16:
@dimaxano that won't allow openvslam from picking up frames over the network and perform localization 🔢

angussmitchell, 2020-05-19:
Any movement on this? My thoughts were that all of the example scripts seem to try to either include pangolin or socketViewer, so if you built without these, you'd fail to run any of the examples

dimaxano, 2020-05-19:
I've built ROS part of OpenVSLAM w/  `-DUSE_PANGOLIN_VIEWER=OFF -DUSE_SOCKET_PUBLISHER=OFF` and it works OK.

angussmitchell, 2020-05-20:
Oh great, thanks for the update. I'll try on the desktop install today