xdspacelab/openvslam #318: Localization + Mapping

CameronFraser, 2020-05-15:
I'm curious on some strategies for making the localization with the mapping flag more robust. It seems very dependent on the original map already being well mapped out and my use case is more that I want to build up a map using multiple videos. A common scenario that happens when I try to run the run_video_localization.cc code is that it won't localize at first but at some point it localizes and then starts mapping. The behavior that I think would be more useful for me is if it can't localize it would start building a new map and once it can localize it reorients the geometry and scale of whatever it was mapping to fit the existing map. I'm a SLAM newbie so I'm not even sure if that is possible or how to approach that. Thoughts? My main goal is to build up a map with anywhere from 5-10 videos taken over a few months period of time that gets more accurate as more videos for that area are fed to it. Any suggestions would be greatly appreciated

mirellameelo, 2020-05-15:
@CameronFraser I`m confusing, are you focused on collaborative SLAM or place recognition? If I understood right, your application will always start mapping until it notices this is a place already mapped and then fusion the point clouds, and then act in localization mode. Could you share some practical application of this feature?

CameronFraser, 2020-05-15:
I don't think it's collaborative SLAM, the scenario is that a robot with a camera goes through an area and does mapping on the area, then the next day it goes through the same area again but not necessarily the same path or the same start and end locations and it keeps doing this everyday for a week or two as an example. The first time I run OpenVSLAM I use the `run_video_slam.cc` code and it maps the area and I create a msgpack file. The second time I run OpenVSLAM i use the `run_video_localization.cc` with the --mapping flag and passing the msgpack file from the day before as a map. However it does not seem to consistently localize to the original map even though they are the same area. The SLAM processing is not being done in real time on the robot, it is post processed afterwards so if it doesn't localize there doesn't seem to be anything I can do to fix it. Maybe I'm missing something. I was thinking that it would map until it can localize and then merge the maps or fuse the point clouds or something to that effect, but it doesn't seem to work that way.

mirellameelo, 2020-05-15:
I see... In a perfect scenario of the "localization mode", once given a single image and a map, the system should be able to localize the camera's pose. As far as I know, both Openvslam and ORBSLAM2 use the "bag of words" technique for recognition. It still has some points to improve, but you could also bet in other techniques... With the information you said, I think you could make improvements in the place recognition module of Openvslam and avoid problems instead of fixing them after they happen. 

Other points of view would be nice!