xdspacelab/openvslam #319: random_device: rdseed failed

jnaor, 2020-05-18:
Hello,

After a few seconds of running I get an runtime_error with the message "random_device: rdseed failed". If I'm not mistaken this is thrown as a result of the line:
    auto random_engine = create_random_engine();
in openvslam/util/random_array.cc.

The exception seems to be thrown at random (no pun intended) and is hard to reproduce. It happened without any camera movement.
I'd be grateful for any help.

Thank you,

Jonathan

jnaor, 2020-05-18:
It seems I can avoid the error if I change
`std::random_device random_device;`
to:
`std::random_device random_device("rdrand")`;

in function `create_random_engine` in `openvslam/util/random_array.cc`

Should this have any noticeable effect on the results?

ymd-stella, 2020-05-26:
I found a similar problem.
https://github.com/googleapis/google-cloud-cpp-common/pull/208