xdspacelab/openvslam #321: getting landmark confidence

famattsson, 2020-05-18:
Is there a way to extract some sort of confidence value from the landmarks. Would get_observed_ratio() work?

shinsumicco, 2020-05-19:
Yes. When the observed ratio (= #observed / #observable) is closer to 1.0, the landmark seems more confident.
More mathematically, you can use an error ellipse of each landmark as an indicator of confidence, which can be computed after BA, (and which is not implemented yet in OpenVSLAM) .

dimaxano, 2020-05-20:
Hi, @shinsumicco! Can you please describe what #observed and #observable means in landmarks class?

ymd-stella, 2020-05-26:
@dimaxano See https://github.com/xdspacelab/openvslam/blob/4b18d9a17caf24613b1a5cbe6a729bd5cf0e7a5b/src/openvslam/tracking_module.cc#L362-L363
https://github.com/xdspacelab/openvslam/blob/4b18d9a17caf24613b1a5cbe6a729bd5cf0e7a5b/src/openvslam/tracking_module.cc#L569-L570
https://github.com/xdspacelab/openvslam/blob/4b18d9a17caf24613b1a5cbe6a729bd5cf0e7a5b/src/openvslam/tracking_module.cc#L541-L542

and [frame::can_observe](https://github.com/xdspacelab/openvslam/blob/fa7b59ec9d59b34b203e3c64a0ef7916e5628dfc/src/openvslam/data/frame.cc#L172).