xdspacelab/openvslam #323: SLAM System reset BUG

SKioresku, 2020-05-19:
[image]

I am find up, that that sign can be incorrect.

dimaxano, 2020-05-19:
What sign do you think should be there?

shinsumicco, 2020-05-19:
The conditions of the `if` statement control a reset of the system when tracking is lost within `init_retry_thr` (sec/secs) after the initialization.
So the second condition is correct because **(CURRENT_FRAME_ID - INITIAL_FRAME_ID) / FPS** equals to the elapsed time after the initialization except when frames are skipped.

SKioresku, 2020-05-20:
Now in the developer code system going to reset instantly after Lost state. You can test that.

SKioresku, 2020-05-20:
> What sign do you think should be there?

curr_frm_.id_ - initializer_.get_initial_frame_id() > camera_->fps_ * init_retry_thr

dimaxano, 2020-05-22:
Well, according to your variant system should reset if tracking is LOST and it passed more than `init_retry_thr` seconds from tracking start
Am I right?

SKioresku, 2020-05-24:
> Well, according to your variant system should reset if tracking is LOST and it passed more than `init_retry_thr` seconds from tracking start
> Am I right?

Stop... Now i see that reset system goes if passed less then 5.0 sec after INITIALIZATION (not from the last succed tracking). Now it makes sence.

closed 2020-05-24 by SKioresku