xdspacelab/openvslam #324: Running in windows

dibet, 2020-05-19:
Hi, After a hard work i can run the openvslam over windows 10. I compile it using mingw for 64 bits.
The result obtained is that the SLAM is very slow compared with LInux.
Any solution for this problem.

Zumbalamambo, 2020-05-19:
@dibet Unfortunately it is slow in windows. :(

shinsumicco, 2020-05-20:
Is the BLAS for g2o on your minGW environment optimized for the architecture you are using?

dibet, 2020-05-20:
ups, i don't have BLAS installed, i will compile https://github.com/xianyi/OpenBLAS. After i will supply feedback, thanks.

dibet, 2020-05-21:
Hi
I compile openblas and after recompile g2o and openvslam. The result is the same, maybe iam making something wrong. Can you suppy some additional information related with blas and mingw?
Thanks.
Dibet

shinsumicco, 2020-06-01:
@dibet Sorry for the late reply.
Could you try [this scripts](https://github.com/shinsumicco/openvslam-on-msvc) to build OpenVSLAM with MSVC?
And please compare processing speed between two executables built with MSVC or minGW.
If the executable with MSVC is faster than the one with minGW, it perhaps contains a problem with a compiler option or libraries on minGW environment.
If there are not much difference between the two executables, it seems that a CPU on your computer is not sufficient for running OpenVSLAM.

dibet, 2020-06-03:
Hi, sorry for my delay in the answer, i compile it with openblas as you say and i resolve the problem. It is fast as in linux. I don't have MSVC installed.
Thanks.

closed 2020-06-21 by dibet