xdspacelab/openvslam #325: run_camera_slam error using realsense r200

chyohoo, 2020-05-21:
Hi, I try to use the rgb camera on realsense r200 to provide the input frames. But the frame and mapper viewer is black and got this error:
> Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.
> fine
> QObject::startTimer: Timers cannot be started from another thread

jagauthier, 2020-05-22:
I'm also going to be using a RealSense R200.  What is your platform?   I am using an Intel Aero, and here's what I had to do: (Assumes you're using Debian/Ubuntu ROS)

https://github.com/IntelRealSense/librealsense/tree/legacy

Make sure you check out the legacy branch:
`git clone --branch legacy https://github.com/IntelRealSense/librealsense.git`
Follow the instructions build it. 

You can then launch the ROS package:
`roslaunch realsense_camera r200_nodelet_default.launch `

And it will publish all the camera info.

chyohoo, 2020-05-23:
> I'm also going to be using a RealSense R200. What is your platform? I am using an Intel Aero, and here's what I had to do: (Assumes you're using Debian/Ubuntu ROS) [...]

Thanks! I am not using the ROS but I have solved the problem. Thanks again.

closed 2020-05-23 by chyohoo