xdspacelab/openvslam #326: Real-Time localization

OmarLaz97, 2020-05-22:
Thank you for your work !!
Is there any methods available to run image localization with real time frames ?

ymd-stella, 2020-05-26:
You can use the ROS wrapper of openvslam (run_slam, run_localization) to run with real time frames.
See https://openvslam.readthedocs.io/en/master/ros_package.html

Zumbalamambo, 2020-06-06:
@OmarLaz97 what do you mean by realtime frames?

rebellojason7, 2020-06-19:
I think he means at frame rate ? I tried running the ros versions in localization mode and 20Hz images, but it doesnt seem to keep up with frame rate ? Any thoughts ?

YujiElfahkrany, 2020-06-24:
@rebellojason7 did you try reducing the image size?

rebellojason7, 2020-06-24:
@YujiElfahkrany Thanks for the suggestion, but I am using the same image size for slam and localization, so I dont think the image size makes a difference. Also my image size is 752x480. What I did notice is that there is far more cpu usage when running localization only as opposed to slam. I havent gone through the code but cant imaging why localization would use so much more than slam 

SLAM usage 
[image]

Localization usage
[image]

YujiElfahkrany, 2020-06-25:
@rebellojason7 yes I also noticed this issue, it converges tho, after sometime. 
I think its stuck in some kind of pose optimization

please check this issue for reference  
https://github.com/xdspacelab/openvslam/issues/338

rebellojason7, 2020-06-26:
@YujiElfahkrany Thanks for pointing me to the thread. Any resolution on it ?

rebellojason7, 2020-07-01:
@YujiElfahkrany any luck with this issue ?

YujiElfahkrany, 2020-07-01:
@rebellojason7 for me I launched pose.optimize() from [here](https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/optimize/pose_optimizer.cc#L97) in a different thread and interrupted that thread if it took too much time. 
for me that was a temporary fix

YujiElfahkrany, 2020-07-01:
I think Sumikura san is testing it here #338