xdspacelab/openvslam #328: No map being created

jagauthier, 2020-05-25:
Greetings,

  Running SLAM through ROS.
`rosrun openvslam run_slam_custom -v orb_vocab.dbow2  -c TUM_RGBD_mono_1.yaml`

Pangolin viewer launches with my camera and a blank Map Viewer.  The camera has lots of yellow dots and lines and that's it.  My current environment is lighted pretty well.  When I turn off the light, and there's a small amount of sunlight a map is created.   I don't understand what is happening here.
```

[2020-05-25 17:56:45.361] [I] loading ORB vocabulary: /home/uav122/orb_vocab/orb_vocab.dbow2
[2020-05-25 17:56:46.252] [I] startup SLAM system
[2020-05-25 17:56:46.252] [I] start mapping module
[2020-05-25 17:56:46.252] [I] start global optimization module
[ Turn light off here]
[2020-05-25 18:07:18.561] [I] initialization succeeded with H
[2020-05-25 18:07:18.590] [I] new map created with 52 points: frame 6707 - frame 6708
```

ymd-stella, 2020-05-26:
I think you should post images or a video to discuss it.

YujiElfahkrany, 2020-06-20:
@jagauthier happened with me too, solved by enabling and disabling show_grid and follow camera options and then using the middle button on the mouse in the map viewer to show the map. 

hope this helps ;)