xdspacelab/openvslam #331: No trajectory shown while running KITTI odometry dataset.

Shivani1796, 2020-05-28:
Heyy..
I have two issues regarding running the code.
1) When I run the below one for KITTI datset it show me video window but Pangolin Viewer : Map Viewer don't show the trajectory for the sequence of images.

I have refer : https://openvslam.readthedocs.io/en/master/example.html#kitti-odometry-dataset

2) Also when I use the same for stereo dataset, it shows me error message : 
[image]

[image]

The above message is shown while running stereo dataset.

dimaxano, 2020-05-28:
Hi!
I would suggest to try earlier versions of OpenCV like 3.3.* or 3.4.*. Maybe that is the case

Shivani1796, 2020-05-28:
Thanks for the reply @dimaxano  But I am using the version '3.4". Then too its not working any other possible way to run it.

shinsumicco, 2020-06-02:
@Shivani1796 
Could you try to run OpenVSLAM with OpenCV 4.2.0?
I couldn't reproduce the error with it.

Shivani1796, 2020-06-03:
@shinsumicco Hey, thanks for the response. 
But I do also try with OpenCV 4.2.0 but still it didn't worked for me.