xdspacelab/openvslam #332: run_video_slam alternating between CONSTRUCT: initialize::perspective and DESTRUCT: initialize::perspective

jgochal, 2020-06-01:
Hi - I'm doing a test on OpenVSLAM with video from a monocular camera.  When I do, however, I get an error when it comes to initializing the the perspective camera.  As shown below, there's an infinite alteration between CONSTRUCT: initialize::perspective and DESTRUCT: initialize::perspective

I configured the camera using opencv2.  I'm ran the script on an Ubuntu 16.04 machine that executed the packaged examples without any issue.  As you can see from the below, I'm running it via Docker.

Any thoughts would be much appreciated!

> root@ip-172-31-57-74:/openvslam/build# ./run_video_slam -v ./data/orb_vocab/orb_vocab.dbow2 -m ./data/l1.mp4 -c ./data/config_landscape.yaml --frame-skip 3 --no-sleep --map-db ./data/map.msg --debug
> [2020-06-01 02:50:20.143] [D] CONSTRUCT: config
> [2020-06-01 02:50:20.143] [I] config file loaded: ./data/config_landscape.yaml
> [2020-06-01 02:50:20.143] [D] load camera model type
> [2020-06-01 02:50:20.143] [D] load camera model parameters
> [2020-06-01 02:50:20.144] [D] CONSTRUCT: camera::base
> [2020-06-01 02:50:20.144] [D] CONSTRUCT: camera::perspective
> [2020-06-01 02:50:20.144] [D] compute image bounds
> [2020-06-01 02:50:20.144] [D] load ORB parameters
> [2020-06-01 02:50:20.144] [D] load tracking parameters
> [2020-06-01 02:50:20.144] [D] load depth threshold
> [2020-06-01 02:50:20.144] [D] load depthmap factor
> [2020-06-01 02:50:20.144] [D] CONSTRUCT: system
>   ___               __   _____ _      _   __  __ 
>  / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
> | (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
>  \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
>       |_|                                        
> 
> Copyright (C) 2019,
> National Institute of Advanced Industrial Science and Technology (AIST)
> All rights reserved.
> 
> This is free software,
> and you are welcome to redistribute it under certain conditions.
> See the LICENSE file.
> 
> Camera Configuration:
> - name: tablet
> - setup: Monocular
> - fps: 30
> - cols: 3264
> - rows: 2448
> - color: RGB
> - model: Perspective
>   - fx: 3031.67
>   - fy: 3026.16
>   - cx: 1643.64
>   - cy: 1281.94
>   - k1: -0.0328995
>   - k2: 1.77744
>   - p1: 0.00320916
>   - p2: 0.00280772
>   - k3: -6.76513
>   - min x: -1743.17
>   - max x: 3259.81
>   - min y: 1510.36
>   - max y: 3667.61
> ORB Configuration:
> - number of keypoints: 1000
> - scale factor: 1
> - number of levels: 8
> - initial fast threshold: 20
> - minimum fast threshold: 7
> 
> [2020-06-01 02:50:20.144] [I] loading ORB vocabulary: ./data/orb_vocab/orb_vocab.dbow2
> [2020-06-01 02:50:20.509] [D] CONSTRUCT: data::camera_database
> [2020-06-01 02:50:20.509] [D] CONSTRUCT: data::map_database
> [2020-06-01 02:50:20.509] [D] CONSTRUCT: data::bow_database
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: publish::frame_publisher
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: publish::map_publisher
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: module::initializer
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: module::relocalizer
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: tracking_module
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: mapping_module
> [2020-06-01 02:50:20.510] [D] CONSTRUCT: global_optimization_module
> [2020-06-01 02:50:20.510] [I] startup SLAM system
> [2020-06-01 02:50:20.510] [I] start mapping module
> [2020-06-01 02:50:20.510] [I] start global optimization module
> [2020-06-01 02:50:20] [connect] Successful connection
> [2020-06-01 02:50:20] [connect] WebSocket Connection 127.0.0.1:3000 v-2 "WebSocket++/0.8.1" /socket.io/?EIO=4&transport=websocket&t=1590979820 101
> [2020-06-01 02:50:20.511] [I] connected to server
> [2020-06-01 02:50:20.663] [D] CONSTRUCT: initialize::perspective
> [2020-06-01 02:50:20.743] [D] DESTRUCT: initialize::perspective
> [2020-06-01 02:50:20.821] [D] CONSTRUCT: initialize::perspective
> [2020-06-01 02:50:20.895] [D] DESTRUCT: initialize::perspective
> [2020-06-01 02:50:20.972] [D] CONSTRUCT: initialize::perspective
> [2020-06-01 02:50:21.046] [D] DESTRUCT: initialize::perspective
> [repeats indefinitely until keyboard break]

shinsumicco, 2020-06-01:
@jgochal 
Have you confirmed the proper behavior with the sample dataset?
If you've confirmed it, the sequence you are using is not appropriate for initialization of SLAM system.

jgochal, 2020-06-01:
Yes, the examples run properly.  Is there any way to tell what in particular is making the sequence not appropriate?

jgochal, 2020-06-02:
When I resize the video resolution to 800 x 600, it works.  The results are poor, but that probably has more to do with the monocular RGB video than OpenVSLAM.  

I'll close this issue.

closed 2020-06-02 by jgochal

RashidLadj, 2020-06-30:
Hello Mr. @shinsumicco ,

What do you mean by the video sequence that we use is not suitable for the initialization of SLAM?

on my part I want to run OpenVSLAM on a video stream taken with my phone (Sumsung S7, Camera perspective)

in order to build the config.yaml file, I had to go through OpenCV, I got the calibration parameters and I built my file as follows:

# =============== #
# Camera Model #
# =============== #

Camera.name: "Sumsung S7"
Camera.setup: "monocular"
Camera.model: "perspective"

Camera.fx: 3011.02271342967
Camera.fy: 2988.9147463804147
Camera.cx: 1873.346399154924
Camera.cy: 1154.1336958461982

Camera.k1: 0.31293082944322254
Camera.k2: -0.7331347425137079
Camera.p1: 0.0023037665461641986
Camera.p2: -9.914234265170498e-05
Camera.k3: -0.0687618693126404

Camera.fps: 60.0
Camera.cols: 3840
Camera.rows: 2160

Camera.color_order: "RGB"

# ================= #
# ORB Parameters #
# ================= #

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles:
  - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.84, 1.0]
  - [0.0, 0.2, 0.7, 1.0]
  - [0.8, 1.0, 0.7, 1.0]

I would like to know please why with this data set the map is not built nothing is happening. more if possible I would like to know that it is the impact of the extinsinc parameter (K1, K2, P1, P2, K3) because I saw a lot of examples where they put 0 0 0 0 0 , so I wonder if it is necessary?

I thank you in advance for your response.

Regards,
Rashid