xdspacelab/openvslam #337: crash after reset then enable/disable mapping

YujiElfahkrany, 2020-06-02:
during testing I found this problem: 
after press reset from gui then enable/disable mapping the program crash. 
a video here: 

https://drive.google.com/drive/folders/1JZlNFUG97Qg9U9iVtdQTayG4U8fDG5y7?usp=sharing

shinsumicco, 2020-06-02:
@YujiElfahkrany 
Could you provide us a stack-trace log?

YujiElfahkrany, 2020-06-03:
Thanks for the fast correspondence. 
is this what you are asking? 
[image]

YujiElfahkrany, 2020-06-03:
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
*** Aborted at 1591145803 (unix time) try "date -d @1591145803" if you are using GNU date ***
PC: @     0x7fc516e20e97 gsignal
*** SIGABRT (@0x3e900000daf) received by PID 3503 (TID 0x7fc4e5f6c700) from PID 3503; stack trace: ***
    @     0x7fc519f49890 (unknown)
    @     0x7fc516e20e97 gsignal
    @     0x7fc516e22801 abort
    @     0x7fc517477957 (unknown)
    @     0x7fc51747dae6 (unknown)
    @     0x7fc51747db21 std::terminate()
    @     0x7fc51747dd54 __cxa_throw
    @     0x7fc517479a23 (unknown)
    @     0x7fc51a233e37 openvslam::data::keyframe::get_landmarks()
    @     0x7fc51a27936e openvslam::match::bow_tree::match_frame_and_keyframe()
    @     0x7fc51a291549 openvslam::module::frame_tracker::bow_match_based_track()
    @     0x7fc51a1fec21 openvslam::tracking_module::track_current_frame()
    @     0x7fc51a200503 openvslam::tracking_module::track()
    @     0x7fc51a201184 openvslam::tracking_module::track_monocular_image()
    @     0x7fc51a1f3aec openvslam::system::feed_monocular_frame()
    @     0x556d83b07863 (unknown)
    @     0x7fc5174a86df (unknown)
    @     0x7fc519f3e6db start_thread
    @     0x7fc516f0388f clone
Aborted (core dumped)