xdspacelab/openvslam #338: pose optimization with few keyframes eats all cpu

YujiElfahkrany, 2020-06-03:
Thank you very much for the clean code!

I noticed that in mapping mode and also localization mode, when there are few keyframes, in the beginning of tracking or mapping, pose_optimizer.optimize() eats all cpu, but when the camera is moves.. cpu usage slowly converges and frame tracking time decreases from about 200 ms to usual 30 ms.. 

I tried adding constraint not to optimize until the first few keyframes are produced, this worked on mapping. but re-localization couldn't really work. 

any clues?

YujiElfahkrany, 2020-06-09:
happens more when the camera is not moving, thus no new keyframes

shinsumicco, 2020-06-24:
@YujiElfahkrany Thank you for reporting the issue, but I'm sorry that I could not reproduce it.
Could you tell me the way in detail to reproduce it if possible?

YujiElfahkrany, 2020-06-24:
@shinsumicco thank you for correspondence. 
for me, just running with a live camera in stereo mode with perspective camera made this happen every time. 
have you tried with live camera?
if it doesn't happen for you in live mode, will close this issue.

YujiElfahkrany, 2020-06-24:
here is my stereo image sequence as a rosbag and the corresponding camera calibration file

left image topic /zmp/left_image_bgr8
right image topic /zmp/right_image_bgr8

hope this helps
https://drive.google.com/drive/folders/1zGJy6ogpzlRQZKyYfKvW48rKMIQX3vJs?usp=sharing

rebellojason7, 2020-06-29:
@YujiElfahkrany any resolution on this ?
@shinsumicco did you manage to test it ?
Thanks

shinsumicco, 2020-06-29:
@YujiElfahkrany Thank you. Please let me test it.

rebellojason7, 2020-06-29:
@shinsumicco thanks for testing it. Just as further information these are my cpu loads when running SLAM vs Localization

SLAM:
[image]

Localization:
[image]

rebellojason7, 2020-07-01:
@shinsumicco @YujiElfahkrany After a few more tests I think there is a problem with the threading of the mapping module. Here is what I did.

1. I built a map by using a simulated world in Gazebo. Everything worked fine and had good cpu usage like the SLAM shown above.
2. I ran localization on the saved map with the "--mapping" flag disabled and had the poor cpu usage like shown above.
3. I ran localization on the saved map with the "--mapping" flag enabled and had good cpu usage again.

I conclude that when the mapping thread is disabled thats when it leads to poor cpu usage. I also watched this video https://www.youtube.com/watch?v=kdXGTGveme8&list=PL_dsdStdDXbrzGQUMh2sy6T8GcCCst3Nm&index=6&t=0s
where he talks about concurrency and conditional variables and has an example of cpu usage. I feel a conditional variable needs to be included here somewhere. 

Let me know your thoughts
Best,
Jason

rebellojason7, 2020-07-21:
@shinsumicco Any update on this ?