xdspacelab/openvslam #340: model scale

monschine, 2020-06-04:
Hi, is there any method to get the scale of the model that we get ?
Just like in the teaser image 
[image]

Thanks in advance

YujiElfahkrany, 2020-06-05:
@monschine are you talking about stereo or mono? 

Mono, there is no way we can get scale information without additional sensor. 
for the stereo case, the generated map will be up to scale. 

this answered your question?

Zumbalamambo, 2020-06-05:
@YujiElfahkrany why not possible with mono? Is it not possible to find the scale by exporting the pointcloud to ply file?

YujiElfahkrany, 2020-06-06:
@monschine @Zumbalamambo for more details check this thread, its about orb_slam2 but true for all visual slam algorithms https://github.com/raulmur/ORB_SLAM2/issues/478#issuecomment-347690730

Zumbalamambo, 2020-06-07:
So isn't it possible to use Openvslam for AR?..
Openvslam gives 6dof position of the camera in the world position. I'm more interested to know where exactly those points are in the model space.

Zumbalamambo, 2020-06-07:
@YujiElfahkrany what if I create the map with stereo and localize using mono?

YujiElfahkrany, 2020-06-07:
@Zumbalamambo yes, stereo slam eliminates the problem of scale by having the true depth of each pixel in the scene. However a correct stereo calibration becomes a must. here is how to do it 
https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

Zumbalamambo, 2020-06-07:
Nice!... So localizing using monocular camera while utilizing the stereo camera to create the map shouldn't be a problem. Did I interpret correctly? ☺️

YujiElfahkrany, 2020-06-09:
localizing using a different camera depends on the descriptors robustness, since orb descriptors are used for this project its supposed to be scale and rotation invariant. the problem will be whether or not the descriptor is illumination invariant, this is ongoing research, but if you can match features from different, correctly calibrated cameras, then virtually you can localize using mono in a stereo built map. The more direct way to tackle this problem is to couple mono with IMU, ongoing for this repo, then you have correct scale for maps built in mono too. 

check these for reference: 
https://stackoverflow.com/questions/36186704/how-can-i-use-features-surf-orb-to-detect-object-with-brightness-invariant
https://medium.com/data-breach/introduction-to-orb-oriented-fast-and-rotated-brief-4220e8ec40cf