xdspacelab/openvslam #347: ROS1 pose + odometry publishing

anbello, 2020-06-14:
I added pose and odometry message publish plus transform broadcasting both in run_localization and run_slam in this fork:
https://github.com/anbello/openvslam/tree/pr-ros1-pose-pub/ros/1/src/openvslam/src

In doing this I used the work done by @klintan and @mirellameelo here: https://github.com/xdspacelab/openvslam/pull/225

I hope this could be of help to this project.

samarth8927, 2020-06-28:
I have replaced files in folder src/openslam with the files provided here, but on launching openvslam i am not able to see any topics with regard to pose and odometry. what can be the issue.

anbello, 2020-06-28:
The commits I added on my fork are vs branch develop. You should clone openvslam, checkout branch develop and the add my commits.
You can also clone from my fork and then checkout branch pr-ros1-pose-pub, this should work for sure, from the fact that for me it works.

samarth8927, 2020-06-29:
I cloned the branch pr-ros1-pose-pub and tried building it, the error that I encountered on build was:

CMake Error at 1/src/openvslam/CMakeLists.txt:62 (find_package):
  By not providing "FindGlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Glog", but
  CMake did not find one.

  Could not find a package configuration file provided by "Glog" with any of
  the following names:

    GlogConfig.cmake
    glog-config.cmake

  Add the installation prefix of "Glog" to CMAKE_PREFIX_PATH or set
  "Glog_DIR" to a directory containing one of the above files.  If "Glog"
  provides a separate development package or SDK, be sure it has been
  installed.

I installed the glog directly and even built it from source, by cloning from this link https://github.com/google/glog, and added 

add_executable(myapp main.cpp)
target_link_libraries(myapp glog::glog)

to CMakeLists.txt file but till the above error is persisting

samarth8927, 2020-07-06:
> I cloned the branch pr-ros1-pose-pub and tried building it, the error that I encountered on build was:
>
> CMake Error at 1/src/openvslam/CMakeLists.txt:62 (find_package):
> By not providing [...]

**Can anyone help me with this issue, not able to resolve it**

wbercode, 2020-07-15:
> **Can anyone help me with this issue, not able to resolve it**

Open the file openvslam/ros/1/src/openvslam/CMakeLists.txt and delete a get_filename_component(DEFAULT_OpenVSLAM_DIR ${DEFAULT_OpenVSLAM_DIR} DIRECTORY).

anbello, 2020-07-15:
I don't have that error, maybe because I installed glog from source https://github.com/google/glog

samarth8927, 2020-07-26:
@wbercode I am no longer getting the glog error after removing get_filename_component(DEFAULT_OpenVSLAM_DIR ${DEFAULT_OpenVSLAM_DIR} DIRECTORY). But i am getting new error:

The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Pangolin_Viewer_LIB
    linked by target "run_localization" in directory /home/sam_pc/em/openvslam/ros/1/src/openvslam/src
    linked by target "run_slam" in directory /home/sam_pc/em/openvslam/ros/1/src/openvslam/src
I have already installed pangolin, but still the error is coming, any suggestions with this regard.

anbello, 2020-07-26:
Have you followed the wiki to build for ROS?
https://openvslam.readthedocs.io/en/master/ros_package.html

samarth8927, 2020-07-27:
@anbello yes I have followed the procedure as given in https://openvslam.readthedocs.io/en/master/ros_package.html, I have downloaded the dependencies as given in link as well as followed the build procedure as mentioned in it, but still getting this error
The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Pangolin_Viewer_LIB
linked by target "run_localization" in directory /home/sam_pc/em/openvslam/ros/1/src/openvslam/src
linked by target "run_slam" in directory /home/sam_pc/em/openvslam/ros/1/src/openvslam/src
I have already installed pangolin, but still the error is coming, any suggestions with this regard.

wbercode, 2020-07-28:
> @anbello yes I have followed the procedure as given in https://openvslam.readthedocs.io/en/master/ros_package.html, I have downloaded the dependencies as given in link as well as followed the build [...]

I guess your previous build mode was SocketViewer. You can continue to maintain this mode, or delete the build folder and re-build using PangolinViewer mode.

FavoRiteKK, 2020-09-10:
Hello,

I got some issue regarding camera pose + odometry.

Environment: ros melodic, BUILD_WITH_MARCH_NATIVE=OFF.

The pose and odometry for some reason, jumps its rotation so quick. 
[image]
[image]

I expect the camera's odometry only rotates around Y axis.

Can anyone help me with this?

PauloRodriguesJr, 2020-09-13:
Hello,

I'm trying to run slam and localization using the mentioned fork, but the performance of the algorithm has dropped dramatically when running from the ROS nodes.
The results i'm getting when i run the code without ROS are still good, but  i am not managing to make the results from ROS do well. 

Do you know what can be the issue? I've tried to run the example videos for debugging, and published the video from a node with 120 FPS, as well as testing different frameskipping options .

Thanks in advance.