xdspacelab/openvslam #348: SLAM with stereo camera (Intel Realsense D435)

Svinterud, 2020-06-15:
Hi, 

I am running ROS2 Dashing and I just installed Openvslam, using the ReadMe.txt provided in #247.
Ive tried to run the "run_slam" node using the provided stereo config file, but the process terminates as a result of "Invalid setup type: Stereo". 

Is stereo supported? If so, how to I run SLAM using a stereo camera?

Thanks in advance!

ymd-stella, 2020-06-15:
> Is stereo supported?

Not suppoted for ros2.
https://github.com/xdspacelab/openvslam/pull/276 is supporting for ros1/stereo, but not merged.

If you want to create ros2/stereo support, [my working branch](https://github.com/ymd-stella/openvslam/commits/feat/ros2-stereo) may be helpful.

dinarkino, 2020-07-10:
I have the same problem with Openvslam not in ROS, but just in docker. I tried to run Kaist dataset in monocular mode using run_image_slam and seems it works fine, but when I try stereo mode I'm getting "Invalid setup type: Stereo error."

Is stereo supported in run_image_slam or I should I use run_kitti_slam or run_euroc_slam where the stereo mode is definitely working?

ymd-stella, 2020-08-07:
https://github.com/xdspacelab/openvslam/blob/6de5fec058647297ca3458e4719b42947fd4a582/example/run_image_slam.cc#L206-L213

mirellameelo, 2020-08-13:
@ymd-stella is your branch "ros2/stereo" working? If it doesn't, could please specify how can I help with that? tks!

ymd-stella, 2020-08-15:
> is your branch "ros2/stereo" working?

Only mono is implemented.
I think you can implement `ros/2/src/openvslam/src/openvslam_ros.cc` with reference to `ros/1/src/openvslam/src/openvslam_ros.cc`.