xdspacelab/openvslam #349: bad results with ros version

rebellojason7, 2020-06-15:
Hi,

Thanks for the good code and making it freely available. I tried running the ROS slam system by streaming images in Gazebo. (ORB slam works great when I do this). However, I find that openv slam cant seem to initialize or loses tracking after 5 seconds from initialization. 

Another aspect, I downloded both versions of EuRoC dataset. When I do ./run_euroc_slam on the dataset (images) it works great and can create the map. However if I run the euroc bag and run the slam system from ros (run_slam) it cannot initialize. 

I also noticed that when I run it from the (run_euroc_slam), it seems like it is not processing it at frame rate. I check this by also playing the bag in the background and can clearly see that the bag euroc dataset is running faster than the images handled by run_euroc_slam.

Any thoughts ?

rebellojason7, 2020-06-15:
I also tested this with the V1_01_easy data and see the same issues where it works with the images in the dataset but not the bag

dimaxano, 2020-06-16:
Hi! What command do you use to run ros version of open-vslam?

rebellojason7, 2020-06-16:
rosrun openvslam run_slam -v orb_vocab.dbow2 -c own_ros.yaml
(own_ros.yaml) is my configuration file for the parameters

rebellojason7, 2020-06-17:
An update with regard to testing openvslam. I have a ximea camera and decided to test it in the real world as opposed to simulation.
I collected a bag from my ximea camera and converted that bag into two formats (video .mp4 and pngs)
For the bag, I ran (run_slam) and openvslam could not initialize
For the .mp4, I ran (run_video_slam) and again it could not initialize
For the pngs, I ran the (run_euroc_slam) which following the same euroc folder format with my data and it was able to build a map.
My conclusions, openvslam seems to work only with a set of images (not .mp4 and bag files) atleast for the perspective camera model.
Any updates or thoughts would be appreciated.

dimaxano, 2020-06-17:
Hmm... 
Very interesting, can you run ros version with `--debug` flag
And show the output

rebellojason7, 2020-06-18:
jrebello@jrebello:~/projects/openvslam/ros/src/openvslam/src$ rosrun openvslam run_slam -v ../data/orb_vocab.dbow2 -c ../data/ximea_ros.yaml --debug
[2020-06-17 23:19:33.988] [D] CONSTRUCT: config
[2020-06-17 23:19:33.988] [I] config file loaded: ../data/ximea_ros.yaml
[2020-06-17 23:19:33.988] [D] load camera model type
[2020-06-17 23:19:33.988] [D] load camera model parameters
[2020-06-17 23:19:33.988] [D] CONSTRUCT: camera::base
[2020-06-17 23:19:33.988] [D] CONSTRUCT: camera::perspective
[2020-06-17 23:19:33.988] [D] compute image bounds
[2020-06-17 23:19:33.988] [D] load ORB parameters
[2020-06-17 23:19:33.988] [D] load tracking parameters
[2020-06-17 23:19:33.988] [D] load depth threshold
[2020-06-17 23:19:33.988] [D] load depthmap factor
[2020-06-17 23:19:33.988] [D] CONSTRUCT: system
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: test
- setup: Monocular
- fps: 20
- cols: 1280
- rows: 1024
- color: Gray
- model: Perspective
  - fx: 658.255
  - fy: 657.755
  - cx: 641.886
  - cy: 508.296
  - k1: -0.1899
  - k2: 0.0556741
  - p1: 0.000434906
  - p2: 0.000112995
  - k3: 0
  - min x: -115.04
  - max x: 1394.21
  - min y: -91.8439
  - max y: 1114.24
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7

[2020-06-17 23:19:33.988] [I] loading ORB vocabulary: ../data/orb_vocab.dbow2
[2020-06-17 23:19:34.371] [D] CONSTRUCT: data::camera_database
[2020-06-17 23:19:34.371] [D] CONSTRUCT: data::map_database
[2020-06-17 23:19:34.371] [D] CONSTRUCT: data::bow_database
[2020-06-17 23:19:34.372] [D] CONSTRUCT: publish::frame_publisher
[2020-06-17 23:19:34.372] [D] CONSTRUCT: publish::map_publisher
[2020-06-17 23:19:34.372] [D] CONSTRUCT: module::initializer
[2020-06-17 23:19:34.372] [D] CONSTRUCT: module::relocalizer
[2020-06-17 23:19:34.372] [D] CONSTRUCT: tracking_module
[2020-06-17 23:19:34.372] [D] CONSTRUCT: mapping_module
[2020-06-17 23:19:34.372] [D] CONSTRUCT: global_optimization_module
[2020-06-17 23:19:34.372] [I] startup SLAM system
[2020-06-17 23:19:34.372] [I] start global optimization module
[2020-06-17 23:19:34.372] [I] start mapping module
[2020-06-17 23:19:40.733] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:42.471] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:42.538] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:42.828] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:42.890] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:43.036] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:43.099] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:43.611] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:43.675] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:43.967] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:44.030] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:44.497] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:44.561] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:45.396] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:45.460] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:46.675] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:46.739] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:48.760] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:48.826] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:49.277] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:49.343] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:49.640] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:49.708] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:50.610] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:50.677] [D] CONSTRUCT: initialize::perspective
[2020-06-17 23:19:50.817] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:50.879] [D] CONSTRUCT: initialize::perspective
^C^C[2020-06-17 23:19:52.896] [I] terminate global optimization module
[2020-06-17 23:19:52.896] [I] terminate mapping module
[2020-06-17 23:19:52.898] [I] shutdown SLAM system
median tracking time: 0.0750755[s]
mean tracking time: 0.0746115[s]
[2020-06-17 23:19:52.899] [D] DESTRUCT: global_optimization_module
[2020-06-17 23:19:52.899] [D] DESTRUCT: mapping_module
[2020-06-17 23:19:52.899] [D] DESTRUCT: tracking_module
[2020-06-17 23:19:52.899] [D] DESTRUCT: module::relocalizer
[2020-06-17 23:19:52.899] [D] DESTRUCT: module::initializer
[2020-06-17 23:19:52.899] [D] DESTRUCT: initialize::perspective
[2020-06-17 23:19:52.899] [I] clear BoW database
[2020-06-17 23:19:52.899] [D] DESTRUCT: data::bow_database
[2020-06-17 23:19:52.899] [I] clear map database
[2020-06-17 23:19:52.899] [D] DESTRUCT: data::map_database
[2020-06-17 23:19:52.899] [D] DESTRUCT: data::camera_database
^C[2020-06-17 23:19:53.013] [D] DESTRUCT: system
[2020-06-17 23:19:53.014] [D] DESTRUCT: publish::map_publisher
[2020-06-17 23:19:53.014] [D] DESTRUCT: publish::frame_publisher
[2020-06-17 23:19:53.073] [D] DESTRUCT: camera::perspective
[2020-06-17 23:19:53.073] [D] DESTRUCT: camera::base
[2020-06-17 23:19:53.073] [D] DESTRUCT: config

rebellojason7, 2020-06-18:
Here are a few images 
[image]

[image]

rebellojason7, 2020-06-18:
and this is with running it after converting the bag to a sequence of images and timestamps. Here it works fine
[image]
[image]

dimaxano, 2020-06-18:
Can you try to change `bgr8` in this [line](https://github.com/xdspacelab/openvslam/blob/dd8af1cfda9ca112d171f601a87b3af803260708/ros/src/openvslam/src/run_slam.cc#L62) to `mono8` ?

rebellojason7, 2020-06-18:
Ahhh, you're great. It worked well. I should have checked the code but thought having fed in "Gray" in the config file it would have automatically selected mono8 instead of bgr8. I should change the code to use the parameter from the config file

closed 2020-06-18 by rebellojason7

dinarkino, 2020-07-29:
Seems that I have exactly the same issue. I'm running openvslam with ros inside docker. I tried to replace bgr8 to mono8 for euroc and nothing happened, also I tried colour images and still seems like images go very fast and the algorithm can't initialise. Sometimes it succeeds but just for a second. Before that, I used openvslam inside docker but without ros and everything worked well