xdspacelab/openvslam #352: pose jumps in y

YujiElfahkrany, 2020-06-17:
this is the output of save_frame_trajectory()
I am wondering if this is normal behavior 
[image]

dimaxano, 2020-06-18:
+1, also noticed this behavior
@shinsumicco Any ideas?

shinsumicco, 2020-06-18:
Thank you for reporting the bug, @YujiElfahkrany, @Zumbalamambo, and @dimaxano .
Is the similar graph draw for `keyframe_trajectory.txt` ?

dimaxano, 2020-06-18:
@shinsumicco For me not, `keyframe_trajectory.txt` poses are good.
Also would like to note, that i see jumps as in x, y, z position coordinates, as in x, y, z, w quaternion rotation values
Here is my `frame_trajectory.txt` file for analyzing if needed
[frame_trajectory.txt](https://github.com/xdspacelab/openvslam/files/4798477/frame_trajectory.txt)

YujiElfahkrany, 2020-06-18:
@shinsumicco the jumps are usually just opposite sign, I suspect some conversion problem from eigen library?

shinsumicco, 2020-06-20:
Thank you for providing me the information, @dimaxano, @YujiElfahkrany .
Please give me a time to inspect it.

shinsumicco, 2020-06-20:
Could you tell me the way to reproduce it?

YujiElfahkrany, 2020-06-23:
@shinsumicco I used gazebo stereo_camera plugin to feed stereo to openvslam using feed_stereo_frame()

the robot on which the camera is attached was going went into the same street twice. 
I can provide image sequence, will need a little time.

dinarkino, 2020-06-23:
Seems like the same problem here. I used Euroc Machine Hall 05 sequence in stereo mode. Here is ape from evo for frame trajectory and keyframe trajectory. Result for keyframe trajectory looks good

[image]
[image]

[image][image]

dinarkino, 2020-07-03:
Just checked and now trajectory for Euroc Machine Hall 05 looks perfect. This time I run the openvslam using docker

[image]

YujiElfahkrany, 2020-07-03:
Interesting, did you try live data?