xdspacelab/openvslam #353: Scale Drift

Zumbalamambo, 2020-06-17:
The scale drifts with monocular mapping and localization. How can I fix it?

YujiElfahkrany, 2020-06-20:
@Zumbalamambo scale drift in monocular can be fixed with loop-closure. 
So you first map your area in a way loop-closure is triggered. for example: if you want to localize along a street make a map of the whole block so that by the end of the mapping you come back to the same street facing the same way. This should trigger loop-closure (graph optimization) and should eliminate your scale drift. save the map and then re-use it for localization. during localization it is advised ( from my testing experience) to turn off the mapping module as it might build a parallel map with another scale drift. 

hope this helps ;)

Zumbalamambo, 2020-06-21:
@YujiElfahkrany superb explanation! thank you!... How can I use openvslam for indoor navigation? The point cloud is sparse so its hard to say where in the real world the 3d coordinates are present. Do you have any idea? For example, it gives me a coordinate of (0.1,0.3.0.3) but how can I map it to the real world for indoor navigation?

YujiElfahkrany, 2020-06-23:
@Zumbalamambo you can use stereo localization out of the box for navigation. 
check these issues for getting the pose with the world coordinates 
https://github.com/xdspacelab/openvslam/issues/206
https://github.com/xdspacelab/openvslam/issues/58

for mono, you have to fix the unknown scale problem before you use the pose for navigation. 

please refer to this for more information about this 

https://github.com/xdspacelab/openvslam/issues/340

if this is fixed please close the issue :) 
thanks