xdspacelab/openvslam #355: Camera IMU alignment

SKioresku, 2020-06-18:
I have measurements from rotation sensor of mobile phone and i want to rotate SLAM coordinat system with according to rotation sensor data. But these coordinat systems have different directions and order of axis. First camera have identity rotation and i have rotation from sensor.  My formula is NewSlamRotMatr = OldSlamRotMatr * CamImuMatr * SensorRotMatr.  But i think that i doing something wrong((

shinsumicco, 2020-06-20:
**In the world-reference coordinate system**, we can acquire an interpolated rotation for the new frame with:
NewWorld2CamRot = CamToImuRot.T * ImuIntegratedRot * CamToImuRot * OldWorld2CamRot

SKioresku, 2020-06-22:
Thank You very much!

closed 2020-06-24 by shinsumicco