xdspacelab/openvslam #357: Finding (0,0,0) in point cloud with reference to the real world

Zumbalamambo, 2020-06-18:
How do I map the origin in the real world to the origin in the point cloud that I've from monocular msg file?

shinsumicco, 2020-06-24:
What does "the origin in the real world" mean?

Zumbalamambo, 2020-06-24:
@shinsumicco for indoor navigation using slam, I have created the msg file. During localization, I would like to know where exactly the 3d coordinates so I can embed AR objects in the real world.

Zumbalamambo, 2020-06-24:
@shinsumicco for example, if I'm close to the window and I get xyz as [1.2,0.3,0.6], I would like to know how I can use this information to apply 3d arrows in the real world. Since the point cloud is sparse, it is hard to set rules.

shinsumicco, 2020-06-30:
@Zumbalamambo 
Indeed, it is hard to find the correspondence between a sparse point-cloud and the real world object.
One solution is to use a reference object with rich texture, e.g., a calibration board.