xdspacelab/openvslam #358: Problems with visualisation

dinarkino, 2020-06-18:
Hi,
thank you for the project! 
I followed the installation guide as well as the simple tutorial. After trouble shooting I finally ran openvslam. It seems that key-point detection and mapping works good but there is several problems with visualisation in map window. 

1) In camera follow mode I see only black screen. It seems that it couldn't follow the camera. When I turn it off, using the mouse, I can zoom out and look at the map.
2) In tracking and mapping I see how the a is created and new points on the map appear, but when the loop is closed points on the map start to blink at different places of the map
3) And the last thing is that also in tracking and mapping mode I see the blue marker of the current position at the initialisation, but then it disappears. In localisation mode, there is no that marker so that is impossible to understand where is the camera. At the same time features in separate windows seems working.

It seems that algorithms is working but visualisation is wrong. What is the possible reason for that? 

System info:
ubuntu 18.04, pangloin viewer, output of the tracking and mapping mode

dinarkino, 2020-06-23:
I've successfully run openvslam using SocketViewer. Everything looks just perfect. Probably, the solution for the issue is to reinstall Pangolin. Maybe I'll try it later. 

The only difference that I found in comparison with Pangolin is that there is no "Loop detection" option in Controls of SocketViewer

wangyunpeng139, 2020-07-01:
@dinarkino  You can click the `Follow Camera` twice in Pangolin., and that might have some effect

dinarkino, 2020-07-03:
Small update: now tried to run inside docker and there Pangolin works without any problems