xdspacelab/openvslam #359: Covariance on Pose

rebellojason7, 2020-06-18:
Hi all,

any idea how to recover the covariance on the pose of the camera either in slam or localization modes ?
Thanks

shinsumicco, 2020-06-20:
You may use "covariance propagation" for recovering covariance of the estimated pose.
Described in Section 5.2 of Multiple View Geometry.

rebellojason7, 2020-06-20:
I understand the math. My question was more on the lines of does openvslam spit out the covariance on the pose or even the covariance for each map point

YujiElfahkrany, 2020-07-01:
@shinsumicco I checked here 
https://arxiv.org/abs/1910.01122 
but couldn't find anything about covariance propagation)
any clues?

rebellojason7, 2020-07-01:
I think this is more of a g2o problem. You need to use the computeMarginals function on the frame vertex.

closed 2020-07-01 by rebellojason7

raghakkrishnan, 2020-11-17:
How do we use computeMarginals() to compute the covariance?