xdspacelab/openvslam #362: Can you run OpenVSLAM with multiple cameras?

robertklee, 2020-06-20:
The application I'm working on will have 3 camera views. Will I be able to use OpenVSLAM with this, or should I look at other SLAM implementations?

Thanks!

shinsumicco, 2020-06-24:
@robertklee We currently cannot feed multiple-camera views to OpenVSLAM, but plan to implement it in the future.

shinsumicco, 2020-06-24:
@robertklee MultiCol-SLAM may be helpful to you.
https://github.com/urbste/MultiCol-SLAM

robertklee, 2020-06-25:
Hmm, thank you for the response! I'm looking for a solution for multi-camera RGBD SLAM, if that exists. Could I stitch together the angles into a equirectangular view and use OpenVSLAM?

shinsumicco, 2020-06-30:
@robertklee 
OpenVSLAM currently cannot accept equirectangular RGB-D input.
But it may be realized with some modification of source code.