xdspacelab/openvslam #363: Aborted (core dumped) when running KITTI dataset

squaresoft2015, 2020-06-22:
I changed the TUM to KITTI in run_kitti_slam.cc for generation of KITTI format trajectory. The compilation is correct, but when I execute the command, the program start for a while and terminated. Displayed as follows:

[2020-06-22 15:32:56.508] [I] loading ORB vocabulary: orb_vocab/orb_vocab.dbow2
[2020-06-22 15:32:56.757] [I] startup SLAM system
[2020-06-22 15:32:56.757] [I] start mapping module
[2020-06-22 15:32:56.757] [I] start global optimization module
[2020-06-22 15:32:58.849] [I] initialization succeeded with F
double free or corruption (out)
*** Aborted at 1592811178 (unix time) try "date -d @1592811178" if you are using GNU date ***
PC: @     0x7ff84f30f3eb gsignal
*** SIGABRT (@0x3e800001584) received by PID 5508 (TID 0x7ff82b6bf700) from PID 5508; stack trace: ***
    @     0x7ff85031c641 (unknown)
    @     0x7ff85006d540 (unknown)
    @     0x7ff84f30f3eb gsignal
    @     0x7ff84f2ee899 abort
    @     0x7ff84f35938e (unknown)
    @     0x7ff84f3614dc (unknown)
    @     0x7ff84f363170 (unknown)
    @     0x7ff84e78b4dd g2o::JacobianWorkspace::~JacobianWorkspace()
    @     0x7ff8501e832c g2o::BlockSolver<>::buildSystem()
    @     0x7ff84e786cc9 g2o::OptimizationAlgorithmLevenberg::solve()
    @     0x7ff84e77f4ac g2o::SparseOptimizer::optimize()
    @     0x7ff850229299 openvslam::optimize::global_bundle_adjuster::optimize()
    @     0x7ff8501d503c openvslam::module::initializer::create_map_for_monocular()
    @     0x7ff8501d559b openvslam::module::initializer::initialize()
    @     0x7ff850143b30 openvslam::tracking_module::initialize()
    @     0x7ff85014636d openvslam::tracking_module::track()
    @     0x7ff850146c12 openvslam::tracking_module::track_monocular_image()
    @     0x7ff850136971 openvslam::system::feed_monocular_frame()
    @     0x55dc7d2ff0df _ZZ13mono_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_jbbbSC_ENKUlvE_clEv
    @     0x7ff84f5aaf74 (unknown)
    @     0x7ff850061669 start_thread
    @     0x7ff84f3eb323 clone
Aborted (core dumped)

Any help?

dimaxano, 2020-06-22:
Hi! Does it work good with TUM trajectory format?

squaresoft2015, 2020-06-22:
It works fine with TUM trajectory format, But I would like to do it in KITTI format.

shinsumicco, 2020-06-24:
@squaresoft2015 Would you try this?
https://openvslam.readthedocs.io/en/master/trouble_shooting.html

squaresoft2015, 2020-06-24:
> @squaresoft2015 Would you try this?
> https://openvslam.readthedocs.io/en/master/trouble_shooting.html

Yes, it works fine now. Thank you! I have changed the TUM to KITTI in run_kitti_slam.cc and recompile the whole project. Now the program output KITTI style trajectory. I want to align the trajectory in evo with plotting parameter -a. But there is no time stamp, so the program output [ERROR] data matrices must have the same shape. How can I solve this problem?

shinsumicco, 2020-06-30:
@squaresoft2015 squaresoft2015
Sorry, I'm not familiar with how to use `evo` tools.
But, we often evaluate OpenVSLAM using the GT trajectory of KITTI in **TUM format**, which is converted from KITTI format using `evo`.