xdspacelab/openvslam #364: fatal error: sio_client.h: No such file or directory

dinarkino, 2020-06-22:
After launching openvslam with Pangolin and facing problems with visualisation #358  I wanted to try it with SocketViewer. I followed the Installation guide but during the final building encountered the error. What am I doing wrong?

cmake output:

```
>     -DBUILD_WITH_MARCH_NATIVE=OFF \
>     -DUSE_PANGOLIN_VIEWER=OFF \
>     -DUSE_SOCKET_PUBLISHER=ON \
>     -DUSE_STACK_TRACE_LOGGER=ON \
>     -DBOW_FRAMEWORK=DBoW2 \
>     -DBUILD_TESTS=ON \
>     ..
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): DISABLED
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Eigen3: /home/dinarkino/depend/include/eigen3 (Required is at least version "3.3") 
-- Use OpenCV 3.4.0
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "5.1.2") 
-- OpenMP: DISABLED
-- SSE3 for ORB extraction (-msse3): DISABLED
-- SSE for floating-point operation (-mfpmath=sse): DISABLED
-- BoW framework: DBoW2 (found in /usr/local/include)
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Found protoc executable: /usr/bin/protoc
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Glog: /usr/include  
-- Stack trace logger: ENABLED
-- Google Perftools: DISABLED
-- Viewer for examples: SocketPublisher
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dinarkino/openvslam/build/googletest-download
Scanning dependencies of target googletest-download
[ 11%] Creating directories for 'googletest-download'
[ 22%] Performing download step (download, verify and extract) for 'googletest-download'
-- Downloading...
   dst='/home/dinarkino/openvslam/build/googletest-download/googletest-download-prefix/src/release-1.10.0.tar.gz'
   timeout='none'
-- Using src='https://github.com/google/googletest/archive/release-1.10.0.tar.gz'
-- verifying file...
       file='/home/dinarkino/openvslam/build/googletest-download/googletest-download-prefix/src/release-1.10.0.tar.gz'
-- Downloading... done
-- extracting...
     src='/home/dinarkino/openvslam/build/googletest-download/googletest-download-prefix/src/release-1.10.0.tar.gz'
     dst='/home/dinarkino/openvslam/build/googletest-src'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 33%] No patch step for 'googletest-download'
[ 44%] No update step for 'googletest-download'
[ 55%] No configure step for 'googletest-download'
[ 66%] No build step for 'googletest-download'
[ 77%] No install step for 'googletest-download'
[ 88%] No test step for 'googletest-download'
[100%] Completed 'googletest-download'
[100%] Built target googletest-download
-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dinarkino/openvslam/build
```

An error:

```
[ 61%] Linking CXX shared library ../../lib/libopenvslam.so
[ 61%] Built target openvslam
Scanning dependencies of target test_openvslam_solve_essential_solver
[ 61%] Running C++ protocol buffer compiler on protobuf/map_segment.proto
Scanning dependencies of target socket_publisher
[ 62%] Building CXX object test/CMakeFiles/test_openvslam_solve_essential_solver.dir/openvslam/solve/essential_solver.cc.o
[ 63%] Building CXX object src/socket_publisher/CMakeFiles/socket_publisher.dir/data_serializer.cc.o
In file included from /home/dinarkino/openvslam_2/src/socket_publisher/data_serializer.cc:1:0:
/home/dinarkino/openvslam_2/src/socket_publisher/data_serializer.h:9:10: fatal error: sio_client.h: No such file or directory
 #include <sio_client.h>
          ^~~~~~~~~~~~~~
compilation terminated.
src/socket_publisher/CMakeFiles/socket_publisher.dir/build.make:70: recipe for target 'src/socket_publisher/CMakeFiles/socket_publisher.dir/data_serializer.cc.o' failed
make[2]: *** [src/socket_publisher/CMakeFiles/socket_publisher.dir/data_serializer.cc.o] Error 1
CMakeFiles/Makefile2:448: recipe for target 'src/socket_publisher/CMakeFiles/socket_publisher.dir/all' failed
make[1]: *** [src/socket_publisher/CMakeFiles/socket_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 64%] Linking CXX executable test_openvslam_solve_essential_solver
[ 64%] Built target test_openvslam_solve_essential_solver
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

System info:
ubuntu 18.04

npiasco, 2020-06-23:
I run into the same problem, just change include path:
 - in */home/dinarkino/openvslam_2/src/socket_publisher/data_serializer.h* change `#include <sio_client.h>` by `#include <sioclient/sio_client.h>`.
 - same for *socket_client.h*

dinarkino, 2020-06-23:
@npiasco thank you! It works

PR #366 (2020-06-24): Fix include path according to the modification of sioclient
fix #364
closed 2020-06-24 by shinsumicco