xdspacelab/openvslam #368: How to get proper map using custom video??

YouMinJung, 2020-06-25:
Hi!

I tested own video in openvslam. However, my result is pretty bad.
I only turn approximately 30 degree, but map point is turned 90 degree. If I turned 180 degree, map point turned 360 degree.
Also, LOST happens quite often.

First time, I tested KITTI dataset with weird yaml file. This result was same as above.
But When I used proper yaml file, result is perfect.
So, I think that if yaml file is not proper, LOST and weird turning actions occur frequently.
**Is it right?**

I checked the yaml file in sample directory and dataset of openvslam (= aist_entrance_hall equirectangular, fisheye).
However, structure of each yaml is little bit different.

I referred the ROS and OpenCV calibration document.
What I do to create yaml file is like this,
1. Take a checker board picture with camera that I want to take the video. (I used smartphone camera.)
2. run calibration code (OpenCV code)
3. Input the frame rate, width, height, color, fx, fy, cx, cy, k1, k2, p1, p2, k3 into yaml file.
4. use that yaml file.

My yaml file is like this,

```
Camera.name: "PhoneCam"
Camera.setup: "monocular"
Camera.model: "equirectangular"

Camera.fx: 2.67028848e+03
Camera.fy: 2.60492480e+033
Camera.cx: 3.52845378e+02
Camera.cy: 3.24774136e+02

Camera.k1: 9.22203220e+00
Camera.k2: -3.98121458e+021
Camera.p1: 3.00868879e-01
Camera.p2: 1.27096103e-01
Camera.k3: 1.01099452e+04

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 1080

Camera.color_order: "RGB"

#================#
# orb parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#========================#
# Initializer Parameters #
#========================#

Initializer.scaling_factor: 5.0
```

**Is there something I did wrong?
Can you advise how to get the proper map using own video?
How to set the each parameter properly?**

mirellameelo, 2020-06-25:
The .yalm file is one of the most important things to have the **best** results.
Are you using a video from your phone?

I'll tell you about my experience. 

# 1 First, I ran my own video using a webcam (a Logitech model). I used the ROS package to find the parameters of it. If you decide to follow this way, please refer to #110 .

# 2 Secondly, I needed to use a video filmed on my phone. Then, I used this tutorial [here](https://www.learnopencv.com/camera-calibration-using-opencv/), which honestly, it was much easier! And you can choose between python or C++.
Basically, after having the chessboard printed, you will use various photos of it as input, and the code will do all the work for you, giving the parameters you need. But at this point, **you definitely must pay attention if the photo resolution coincides with the video resolution**! For example, I was using a Galaxy S8 with the HD resolution for video. But HD resolution is not available for pictures, so I had to film the chessboard, then I read the video (w/ OpenCV as well), selected the frames by interleaving them until having something about 50 frames, and then I used them as input.

After achieving the parameters, you can use #110 as an example to build the appropriate config.yalm file for your camera. You will see that te calibration makes a lot of difference!

mirellameelo, 2020-06-25:
@YouMinJung I also used the phone camera and I had great results. Let me share it with you...

I used this tutorial [here](https://www.learnopencv.com/camera-calibration-using-opencv/). You can choose between python or C++.

Basically, after having the chessboard printed, you will use various photos of it as input, and the code will do all the work for you, giving the parameters you need. But at this point, **you definitely must pay attention if the photo resolution coincides with the video resolution**! I suppose this is what happening to you...

For example, I was using a Galaxy S8 with the HD resolution for video. But HD resolution is not available for pictures, so I had to film the chessboard, then I read the video (w/ OpenCV as well), selected the frames by interleaving them until having something about 50 frames, and then I used them as input. 

After achieving the parameters, you can use #110 as an example to build the appropriate config.yalm file for your camera.

YouMinJung, 2020-06-25:
Thank you very much!!
You save me!!

closed 2020-06-29 by YouMinJung