xdspacelab/openvslam #372: openvslam make error

chluck2008, 2020-06-26:
cmake     -DBUILD_WITH_MARCH_NATIVE=ON     -DUSE_PANGOLIN_VIEWER=ON     -DUSE_SOCKET_PUBLISHER=OFF     -DUSE_STACK_TRACE_LOGGER=ON     -DBOW_FRAMEWORK=DBoW2     -DBUILD_TESTS=ON     ..
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Eigen3: /usr/local/include/eigen3 (Required is at least version "3.3") 
-- Use OpenCV 3.4.7
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so   
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- Found BLAS: /usr/lib/libf77blas.so;/usr/lib/libatlas.so  
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.a
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: TRUE (found version "4.2.1") 
-- OpenMP: DISABLED
-- SSE3 for ORB extraction (-msse3): DISABLED
-- SSE for floating-point operation (-mfpmath=sse): DISABLED
-- BoW framework: DBoW2 (found in /usr/local/include)
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Glog: /usr/include  
-- Stack trace logger: ENABLED
-- Google Perftools: DISABLED
-- Viewer for examples: PangolinViewer
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/openvslam/openvslam_ws/src/openvslam/build/googletest-download
Scanning dependencies of target googletest-download
[ 11%] Creating directories for 'googletest-download'
[ 22%] Performing download step (download, verify and extract) for 'googletest-download'
-- Downloading...
   dst='/home/openvslam/openvslam_ws/src/openvslam/build/googletest-download/googletest-download-prefix/src/release-1.10.0.tar.gz'
   timeout='none'
-- Using src='https://github.com/google/googletest/archive/release-1.10.0.tar.gz'
-- verifying file...
       file='/home/openvslam/openvslam_ws/src/openvslam/build/googletest-download/googletest-download-prefix/src/release-1.10.0.tar.gz'
[... 182 lines cut ...]
  an IMPORTED target, or an ALIAS target is missing?

CMake Error at test/CMakeLists.txt:30 (add_executable):
  Target "test_openvslam_util_trigonometric" links to target "Eigen3::Eigen"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

CMake Error at test/CMakeLists.txt:30 (add_executable):
  Target "test_openvslam_util_trigonometric" links to target "Eigen3::Eigen"
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

CMake Error at test/helper/CMakeLists.txt:2 (add_library):
  Target "test_helper" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

CMake Error at src/openvslam/CMakeLists.txt:41 (add_library):
  Target "openvslam" links to target "Eigen3::Eigen" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /home/openvslam/openvslam_ws/src/openvslam/build
help me ,thanks

rebellojason7, 2020-06-26:
Try with -DBUILD_WITH_MARCH_NATIVE=OFF

chluck2008, 2020-06-27:
@rebellojason7  The problem remains the same, thanks

chluck2008, 2020-06-27:
make -j4
/usr/bin/ld: 找不到 -lEigen3::Eigen
collect2: error: ld returned 1 exit status
test/helper/CMakeFiles/test_helper.dir/build.make:114: recipe for target 'test/helper/libtest_helper.so' failed
make[2]: *** [test/helper/libtest_helper.so] Error 1
CMakeFiles/Makefile2:1618: recipe for target 'test/helper/CMakeFiles/test_helper.dir/all' failed
make[1]: *** [test/helper/CMakeFiles/test_helper.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....

chluck2008, 2020-06-27:
We've solved the problem. Thank you. :)
git clone -b yaml-cpp-0.6.2 https://github.com/jbeder/yaml-cpp 
line 174 CMakeLists.txt
set(yaml_cxx_flags "-Wall ${GCC_EXTRA_OPTIONS} -fPIC -pedantic -Wno-long-long ${yaml_cxx_flags}")

closed 2020-06-27 by chluck2008