xdspacelab/openvslam #374: Installation for Windows 10

hetareba, 2020-06-27:
I tried to install it by referring to Mr. Reunanen's explanation.
But I'm new to building software so I can't understand what I have to build with Visual Studio 2017.
Can anyone please tell me step by step?
I know
[image]
 this is a very annoying request, but I need your help.
I have already tried every method...

(What I should do next? :< )

Reference: [https://github.com/reunanen/openvslam-build-vs](url)

shinsumicco, 2020-06-27:
@hetareba Thank you for having the interest on this project.
I'm sorry that we've not actually tried the instructions posted by @reunanen for building OpenVSLAM on Windows environment.
But I provide PowerShell scripts [here](https://github.com/shinsumicco/openvslam-on-msvc) in order to make MSVC project from the CMake scripts.
If you accept this way for your purpose, could you try those scripts?

hetareba, 2020-06-27:
Thanks a lot!!! I'll try it soon, then i'll report the result.

reunanen, 2020-06-27:
@hetareba One way to build a specific project is to use the "Solution Explorer" (see bottom left in the image below), right-click the desired project, and choose "Build".

[image]

hetareba, 2020-06-29:
[image]

Thank you everyone for your cooperation.
The build was successful.
And the attached screenshots show that the tutorial was successful.
As you can see, I ended up using Ubuntu.
But success could not have been possible without your cooperation.
Thank you very much 👍 .

closed 2020-06-29 by hetareba