xdspacelab/openvslam #375: .msg files for Dense construction

belal-ibrahim, 2020-06-27:
I am working on a project that aims to build end -to-end 3D construction
 I have tested openvslam and i could generate map.msg file. However, I need to build dense=>surface=>texture of the scene.
I have referred to #87 but the result wasn't even close to expected, any idea for how to achieve this target?

mirellameelo, 2020-11-06:
@belal-ibrahim For now, you can integrate it with another project. I'm trying to use [RTABMAP](http://wiki.ros.org/rtabmap), for example. Once you input Openvslam's odometry associated with the images (stereo or RGBD) the Rtabmap builds an occupancy grid, dense point cloud, and other interesting things.

belal-ibrahim, 2020-11-07:
@mirellameelo any monocular integration?