xdspacelab/openvslam #378: Tracking map when camera turned

YouMinJung, 2020-06-29:
Hi.

I'm trying to make a map using own video.
I tested several times, but when I turned approximately 90 degree, almost 80% program is LOST.
**It works fine when I walked in a straight line, but when I turns left or right, the tracking is lost.**

I saw the tracking map. 
I think that OpenVSLAM is lost because **failed extract feature** in my case.
Image of ORB tracking result is like below.
[image]

**Right side of image is tracking well. However, left side of image can't track the feature.**

My question is like these.
**1. Is there any way to improve ORB or some other feature extracter?
2. What is the meaning of ORB parameter that openVSLAM used? (in .yaml)**

```
#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 3000
Feature.scale_factor: 1.5
Feature.num_levels: 5
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
```

How to track map when camera turned right/left/360 degree?

shinsumicco, 2020-06-29:
@YouMinJung 
Are you using a monocular camera?
In principle, monocular cameras are NOT robust to pure-rotation because triangulation of 3D points cannot accurately performed with small translation.
If the camera system experiences such movement, please try to use a stereo or an RGB camera instead of a monocular one.

YouMinJung, 2020-06-30:
Even though monocular camera, KITTI dataset (Monocular, Gray) works well.
Is there any different to smartphone camera?

YujiElfahkrany, 2020-07-01:
@YouMinJung  I think offline usage (KITTI dataset, etc.. ) ensures all the frames to be used for tracking, unless we set the skip frame parameter otherwise. However, if you run with live data (video) number of frames skipped depends on the cpu stress of your machine. I see from your picture, your tracking time is 60 ms / frame. 
if you are running with 30 fps, you are skipping about half of the frames which makes the algorithm, especially running with Mono, prone to pure rotation as mentioned above. 
hope this helps ;)

YouMinJung, 2020-07-01:
@YujiElfahkrany 
Thank you for your support.
I have a question.
You said that "if you are running with 30 fps, you are skipping about half of the frames which makes the algorithm, especially running with Mono."
Is there any way to adjust skipping frame except '--frame-skip' parameter?

greymfm, 2020-12-06:
Try to increase the number of used keypoints significantly (for both SLAM and localization):

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 10000   
Feature.scale_factor: 1.1  
Feature.num_levels: 10 
Feature.ini_fast_threshold: 1 
Feature.min_fast_threshold: 0 

Use pre-recorded video sequences (SLAM+localization) for testing. I get improved results with these ORB parameters on them.
The processing performance will be significantly slower, however this can be improved again significantly by CUDA support ( #445).

bot-lin, 2020-12-07:
> @YouMinJung
> Are you using a monocular camera?
> In principle, monocular cameras are NOT robust to pure-rotation because triangulation of 3D points cannot accurately performed with small translation. [...]

Do you mean RGBD in the last sentence?

Thanks