xdspacelab/openvslam #380: Get Localization frame position

hetareba, 2020-06-29:
Thank you for this great job.
I have a question.
I understand that if I specify `--eval-log` when doing **run_video_slam**, trajectory will be recorded.
But I don't know how to record trajectory when doing **Localization**.
I referred to [https://github.com/xdspacelab/openvslam/issues/292#issuecomment-632891805](url), but I thought this is a command that does not support Ubuntu.
**Is it possible to record trajectory only by operating on Ubuntu?**

sumitsarkar1, 2020-07-04:
@hetareba have you found any solution to this ?..I also need to evaluate the trajectory while only in localization mode...plz share if you have any suggestions

YujiElfahkrany, 2020-07-06:
how about calling this function ? 
https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/system.h#L57

sumitsarkar1, 2020-07-07:
@YujiElfahkrany for using openvslam in the ROS environment will the above mentioned solution work too?

sumitsarkar1, 2020-07-08:
in the run_localization.cc :

`    // shutdown the SLAM process
    SLAM.shutdown();
	
	SLAM.save_frame_trajectory("localization_frame_trajectory.txt", "TUM");`

hetareba, 2020-07-09:
> how about calling this function ?
> https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/system.h#L57

Sorry for the late reply!
Many thanks for your comments.

@YujiElfahkrany, thanks for the suggestions.
Do I need to make any changes to **system.h** before building the software in order to call **`void save_frame_trajectory`** from **system.h**?

sorry! I don't understand C++ at all.
Please forgive some elementary questions.

sumitsarkar1, 2020-07-09:
@hetareba no changes in system.h. Just call the function `SLAM.save_frame_trajectory()` . Dont forget to do catkin_make to apply the changes