xdspacelab/openvslam #383: OpenVSLAM is compiled and can be invoked but blocks at the beginning of the sequence

miroslavradojevic, 2020-07-02:
I have compiled openvslam along with all the dependencies as suggested in the [installation](https://openvslam.readthedocs.io/en/master/installation.html). It is possible to invoke:
./run_kitti_slam -h

However, the method stops (freezes) at the very beginning when calling ./run_video_slam and that happens with my own sequences as well as demo sequences provided (such as aist_living_lab_1 or aist_living_lab_2 given in demo examples).

There is no error reported and the application does not crash, just stays idle.

I use Pangolin viewer, and did try to compile using -DBUILD_WITH_MARCH_NATIVE set to both ON and OFF. 
OS: Ubuntu 18.04 (can that be an issue?)
OpenCv 3.4.10 (OpenCV 4.x seems to be not supported yet)
Has anyone had any such experiences? It should be possible to run demos to start with.

YujiElfahkrany, 2020-07-06:
does it eat all cpu?

sudo-panda, 2020-07-11:
@YujiElfahkrany I am facing the same issue but it doesn't eat all the cpu. It just freezes at the first frame.

@miroslavradojevic Did you find a solution?

sudo-panda, 2020-07-11:
This issue is related to #303 but the solution given there didn't work for me.

d-desiatkin, 2020-09-16:
Hi!
Sorry for late response. 
Yeah in my closed issue the reason was with particular build of OpenMP and Jetson architecture.
However later I again meet the same problem. But in that case the error was caused by the wrong calibration of the camera.
I still do not understand that behavior, but please double check the camera calibration, especially if you resize them online (better way is to create separate resized image sequence).
Ensure that your calibration parameters resized accordingly.

sudo-panda, 2020-09-16:
@d-desiatkin Thanks for the reply. Unfortunately I am no longer working on that project so I won't be able to check but someone else might find it useful. :)