xdspacelab/openvslam #384: How to access current tracking state

laxnpander, 2020-07-03:
Hey there,

I am currently testing OpenVSLAM as library for my own project. My problem right now is, how to access basic information about the current tracking state? I feed image data into it, get a pose in return but can't validate if tracking was successful. ORB SLAM2 returned an empty cv::Mat() if tracking has failed. OpenVSLAM however returns an invalid but non-zero eigen matrix even if tracking was not running. 

For keyframes I can use the map publisher in a laborious way of checking whether the number of keyframes has increased since my last run, but it should be straightforward like system->get_tracking_state() to find out what's going on!

Is there a way to achieve what I want to without creating my own OpenVSLAM fork?

Thanks for the info and best regards

YujiElfahkrany, 2020-07-06:
how about this https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/tracking_module.h#L32

laxnpander, 2020-07-06:
@YujiElfahkrany: But how do I get it when creating the slam system? I know I could modify the system.h no problem. But I want to stay compatible with future releases of open vslam. And as far as I can tell there is not getter for the tracker. Only for frame publisher and map publisher.

dimaxano, 2020-07-06:
I also did not find any solution for that and created `get_tracker_state` in system.h and system.cc

laxnpander, 2020-07-06:
@dimaxano: Then this should be a "feature" I'd say. The tracking state is really one of the most basic and important informations for external frameworks using open vslam.

laxnpander, 2020-07-20:
Resulted in a separate PR: [#399 ](https://github.com/xdspacelab/openvslam/pull/399/files)

closed 2020-07-20 by laxnpander