xdspacelab/openvslam #385: Crash in find_via_ransac(...)

laxnpander, 2020-07-03:
Hey there,

another issue popped up for me: I'm processing my own image data. The SLAM system is successfully created and the parameters passed look good. Feeding my first image is not tracking, the second then crashes the application. A GDB backtrace leads me to find_via_ransac(...) causing the trouble. Is there a known issue about it? 

[image]

I am running Ubuntu 18.04 with OpenCV 3.3.1 and Eigen 3.3.4.

Best regards,

Alex

shinsumicco, 2020-07-03:
@laxnpander Thank you for reporting the issue.
Could you paste the stack-trace log as text, not image?

laxnpander, 2020-07-06:
@shinsumicco: Hey, jeah. Sorry xterm somehow doesn't let me copy stuff, so I thought an image might be enough. Here is the copy after getting it to run in a regular terminal:
```
bt
#0  0x00007fffed0c15c7 in Eigen::JacobiSVD<Eigen::Matrix<double, -1, 9, 0, -1, 9>, 2>::compute(Eigen::Matrix<double, -1, 9, 0, -1, 9> const&, unsigned int) ()
    at /usr/local/lib/libopenvslam.so
#1  0x00007fffed0c7f1a in openvslam::solve::fundamental_solver::compute_F_21(std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&) () at /usr/local/lib/libopenvslam.so
#2  0x00007fffed0c8cec in openvslam::solve::fundamental_solver::find_via_ransac(unsigned int, bool) () at /usr/local/lib/libopenvslam.so
#3  0x00007ffff65866df in  () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6a5d6db in start_thread (arg=0x7fff1effd700) at pthread_create.c:463
#5  0x00007ffff5c4388f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

I added some debug information to check what's happening. These are the keypoints and A matrix going into the jacobiSVD function:

```
num_points: 8
i : 0
keypt 1: [0.565133, 1.32746]
keypt 2: [0.45384, 1.34583]
i : 1
keypt 1: [0.444084, -1.96692]
keypt 2: [0.451963, -1.94766]
i : 2
keypt 1: [2.03214, 1.44608]
keypt 2: [2.03702, 1.47658]
i : 3
keypt 1: [-2.14047, 1.78414]
keypt 2: [-2.13962, 1.81751]
i : 4
keypt 1: [0.829777, -0.696307]
keypt 2: [0.831411, -0.698369]
i : 5
keypt 1: [-1.74134, 0.671075]
keypt 2: [-1.7389, 0.696142]
i : 6
keypt 1: [-1.08344, 2.14191]
keypt 2: [-1.09503, 2.16847]
i : 7
keypt 1: [-0.872719, 1.61175]
keypt 2: [-0.860406, 1.6336]
A matrix:
        0         0         0 -0.565133  -1.32746        -1  0.760575   1.78654   1.34583
 0.565133   1.32746         1         0         0         0  -0.25648 -0.602455  -0.45384
        0         0         0 -0.444084   1.96692        -1 -0.864923   3.83089  -1.94766
 0.444084  -1.96692         1         0         0         0  -0.20071  0.888977 -0.451963
        0         0         0  -2.03214  -1.44608        -1   3.00062   2.13527   1.47658
  2.03214   1.44608         1         0         0         0  -4.13951  -2.94571  -2.03702
        0         0         0   2.14047  -1.78414        -1  -3.89033   3.24269   1.81751
 -2.14047   1.78414         1         0         0         0  -4.57981   3.81739   2.13962
        0         0         0 -0.829777  0.696307        -1  -0.57949  0.486279 -0.698369
 0.829777 -0.696307         1         0         0         0 -0.689886  0.578917 -0.831411
        0         0         0   1.74134 -0.671075        -1  -1.21222  0.467163  0.696142
 -1.74134  0.671075         1         0         0         0  -3.02802   1.16693    1.7389
        0         0         0   1.08344  -2.14191        -1   -2.3494   4.64466   2.16847
 -1.08344   2.14191         1         0         0         0   -1.1864   2.34547   1.09503
        0         0         0  0.872719  -1.61175        -1  -1.42568   2.63297    1.6336
-0.872719   1.61175         1         0         0         0 -0.750893   1.38676  0.860406

```

It looks quite fine to me. OpenVSLAM's simple example is also running flawlessly. I assume it's either really some bug, or caused by the fact I use OpenVSLAM as library. I built it just like in the doc, but without any viewer. Installed it and use find_package(openvslam) to link to it.

laxnpander, 2020-07-06:
@shinsumicco: Alright, I think I solved the issue. I found the note in the installation instructions about weird crashes when building with "march=native". So I tried recompiling both g2o and open vslam, now the crash is gone!

closed 2020-07-06 by laxnpander