xdspacelab/openvslam #387: RGBD TUM SLAM crashes when I input my own RGBD data

robertklee, 2020-07-05:
I am trying to input my own RGBD data. I have my setup as `depth/imgxxxx.png` and `rgb/imgxxxx.png` and I have text files rgb.txt and depth.txt with mappings of timing to filename. However, when I run, I get this error:

```
~/Documents/Visual_SLAM/openvslam/build$ ./run_tum_rgbd_slam -v ./orb_vocab/orb_vocab.dbow2 -d ~/Downloads/zed_rgbd_samples/warehouse/proper_format/ -c ~/Downloads/zed_rgbd_samples/ZED_RGBD_rgbd.yaml --frame-skip 3 --no-sleep --map-db map.msg
[2020-07-05 00:00:50.353] [I] config file loaded: /home/rlee/Downloads/zed_rgbd_samples/ZED_RGBD_rgbd.yaml
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not load a timestamp file from /home/rlee/Downloads/zed_rgbd_samples/warehouse/proper_format//rgb.txt
*** Aborted at 1593932450 (unix time) try "date -d @1593932450" if you are using GNU date ***
PC: @     0x7fd26347d428 gsignal
*** SIGABRT (@0x3e8000030f9) received by PID 12537 (TID 0x7fd266995b80) from PID 12537; stack trace: ***
    @     0x7fd265d9b390 (unknown)
    @     0x7fd26347d428 gsignal
    @     0x7fd26347f02a abort
    @     0x7fd263ab784d __gnu_cxx::__verbose_terminate_handler()
    @     0x7fd263ab56b6 (unknown)
    @     0x7fd263ab5701 std::terminate()
    @     0x7fd263ab5919 __cxa_throw
    @           0x43a569 tum_rgbd_sequence::acquire_image_information()
    @           0x43a7ee tum_rgbd_sequence::tum_rgbd_sequence()
    @           0x41a071 rgbd_tracking()
    @           0x41753a main
    @     0x7fd263468830 __libc_start_main
    @           0x417ed9 _start
    @                0x0 (unknown)
Aborted (core dumped)
```

Shashika007, 2020-07-07:
@robertklee your path is incorrect "/Downloads/zed_rgbd_samples/warehouse/proper_format//rgb.txt", Please change "-d ~/Downloads/zed_rgbd_samples/warehouse/proper_format/" into "-d ~/Downloads/zed_rgbd_samples/warehouse/proper_format" without "/" at the end.

robertklee, 2020-07-07:
Thanks for the reply @Shashika007 
I tried it without the "/" at the end and it still didn't work: `./run_tum_rgbd_slam -v ./orb_vocab/orb_vocab.dbow2 -d ~/Downloads/zed_rgbd_samples/warehouse/proper_format -c ~/Downloads/zed_rgbd_samples/ZED_RGBD_rgbd.yaml --frame-skip 3 --no-sleep --map-db map.msg` 

Error:
```
[2020-07-07 00:32:51.818] [I] config file loaded: /home/rlee/Downloads/zed_rgbd_samples/ZED_RGBD_rgbd.yaml
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not load a timestamp file from /home/rlee/Downloads/zed_rgbd_samples/warehouse/proper_format/rgb.txt
*** Aborted at 1594107171 (unix time) try "date -d @1594107171" if you are using GNU date ***
PC: @     0x7f19b9c5f428 gsignal
*** SIGABRT (@0x3e8000037f0) received by PID 14320 (TID 0x7f19bd177b80) from PID 14320; stack trace: ***
    @     0x7f19bc57d390 (unknown)
    @     0x7f19b9c5f428 gsignal
    @     0x7f19b9c6102a abort
    @     0x7f19ba29984d __gnu_cxx::__verbose_terminate_handler()
    @     0x7f19ba2976b6 (unknown)
    @     0x7f19ba297701 std::terminate()
    @     0x7f19ba297919 __cxa_throw
    @           0x43a569 tum_rgbd_sequence::acquire_image_information()
    @           0x43a7ee tum_rgbd_sequence::tum_rgbd_sequence()
    @           0x41a071 rgbd_tracking()
    @           0x41753a main
    @     0x7f19b9c4a830 __libc_start_main
    @           0x417ed9 _start
    @                0x0 (unknown)
Aborted (core dumped)

```

dinarkino, 2020-07-21:
@robertklee make sure that there is folders "depth" and "rgb" as well as files depth.txt and rgb.txt at the **~/Downloads/zed_rgbd_samples/warehouse/proper_format** path. I just tried ./run_tum_rgbd_slam with tum rgbd dataset and other rgbd data. It works. Also check paths inside .txt files, in my case instead of "rgb" folder I have "color" and paths like color/1560004885.446172.png