xdspacelab/openvslam #389: ROS catkin_make error

chluck2008, 2020-07-06:
catkin_make     \
-DBUILD_WITH_MARCH_NATIVE=OFF     \
-DUSE_PANGOLIN_VIEWER=ON    \
 -DUSE_SOCKET_PUBLISHER=OFF     \
-DUSE_STACK_TRACE_LOGGER=ON     \
-DBOW_FRAMEWORK=DBoW2     \
-DUSE_OPENMP=OFF

[100%] Linking CXX executable /home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_slam
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_destroy_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_init_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_unset_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_set_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)’未定义的引用
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_localization.dir/build.make:152: recipe for target '/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_localization' failed
make[2]: *** [/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_localization] Error 1
CMakeFiles/Makefile2:2549: recipe for target 'openvslam/src/CMakeFiles/run_localization.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_localization.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_destroy_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_init_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_unset_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_set_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)’未定义的引用
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_slam.dir/build.make:152: recipe for target '/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_slam' failed
make[2]: *** [/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_slam] Error 1
CMakeFiles/Makefile2:2611: recipe for target 'openvslam/src/CMakeFiles/run_slam.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_slam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

sumitsarkar1, 2020-07-07:
Try the following:
1.) sudo apt remove ros-<distro>-libg2o 
2.) catkin_make -DBUILD_WITH......
3.) sudo apt install ros-<distro>-libg2o  ( <distro> is your ROS distribution like kinetic..melodic...)

chluck2008, 2020-07-07:
@sumitsarkar1  thank you,but I've been following your instructions. It's still the same problem

Shashika007, 2020-07-07:
@chluck2008 try -DBUILD_WITH_MARCH_NATIVE=ON

chluck2008, 2020-07-07:
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_destroy_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_init_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_unset_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_set_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)’未定义的引用
collect2: error: ld returned 1 exit status
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_destroy_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_init_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_unset_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘omp_set_lock’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)’未定义的引用
/home/admin1234/openvslam_ws/src/openvslam/build/lib/libopenvslam.so：对‘g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)’未定义的引用
collect2: error: ld returned 1 exit status
openvslam/src/CMakeFiles/run_localization.dir/build.make:152: recipe for target '/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_localization' failed
make[2]: *** [/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_localization] Error 1
CMakeFiles/Makefile2:2549: recipe for target 'openvslam/src/CMakeFiles/run_localization.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_localization.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
openvslam/src/CMakeFiles/run_slam.dir/build.make:152: recipe for target '/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_slam' failed
make[2]: *** [/home/admin1234/openvslam_ws/src/openvslam/ros/devel/lib/openvslam/run_slam] Error 1
CMakeFiles/Makefile2:2611: recipe for target 'openvslam/src/CMakeFiles/run_slam.dir/all' failed
make[1]: *** [openvslam/src/CMakeFiles/run_slam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed