xdspacelab/openvslam #390: Segmentation fault: 11 Running run_video_slam example on Mac

wbrennan899, 2020-07-06:
OSX 10.15.5 (also had same issue on 10.14)

Any idea where to start looking for a solution?

Note: I built g2o with `-DG2O_USE_OPENMP=OFF` in order to get the build to complete.

`$ ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg --debug`

```
[2020-07-06 12:29:54.884] [D] CONSTRUCT: config
[2020-07-06 12:29:54.885] [I] config file loaded: ./aist_living_lab_1/config.yaml
[2020-07-06 12:29:54.885] [D] load camera model type
*** Aborted at 1594052994 (unix time) try "date -d @1594052994" if you are using GNU date ***
PC: @                0x0 (unknown)
*** SIGSEGV (@0x21) received by PID 90701 (TID 0x11497edc0) stack trace: ***
    @     0x7fff7152e5fd _sigtramp
    @        0x10c9f2fd2 YAML::detail::node::get<>()
    @        0x10ca0fd4d YAML::Node::operator[]<>()
    @        0x10ca10353 openvslam::camera::base::load_model_type()
    @        0x10c9bd3d3 openvslam::config::config()
    @        0x10c9be08d openvslam::config::config()
    @        0x10c888c0c main
Segmentation fault: 11
```

laxnpander, 2020-07-07:
Shouldn't have anything to do with OpenMP. According to the stack trace your config file is causing the segfault. Are you sure there is no typo or anything in the config? Can you post the content here?

wbrennan899, 2020-07-07:
@laxnpander thanks for responding. I'm not sure if there is a typo, I am using the config file downloaded in the first step of the intro tutorial.

https://openvslam.readthedocs.io/en/master/simple_tutorial.html
```
FILE_ID="1d8kADKWBptEqTF7jEVhKatBEdN7g0ikY"
curl -sc /tmp/cookie "https://drive.google.com/uc?export=download&id=${FILE_ID}" > /dev/null
CODE="$(awk '/_warning_/ {print $NF}' /tmp/cookie)"
curl -sLb /tmp/cookie "https://drive.google.com/uc?export=download&confirm=${CODE}&id=${FILE_ID}" -o aist_living_lab_1.zip
unzip aist_living_lab_1.zip
```

Here is the config file (`/aist_living_lab_1/config.yaml`):

```
# Equirectangular model for RICOH THETA S

#==============#
# Camera Model #
#==============#

Camera.name: "RICOH THETA S 960"
Camera.setup: "monocular"
Camera.model: "equirectangular"

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 960

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles:
  - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.84, 1.0]
  - [0.0, 0.2, 0.7, 1.0]
  - [0.8, 1.0, 0.7, 1.0]

```

laxnpander, 2020-07-07:
@wbrennan899: That looks about right. Can you check if your libyaml-cpp-dev is the correct version (>=0.6) and/or if there is more than one version installed?

wbrennan899, 2020-07-07:
@laxnpander what is the best way to check if I have multiple versions of yaml? 

I used `brew install yaml-cpp`
`/usr/local/include/yaml-cpp/` points to `../Cellar/yaml-cpp/0.6.3_1/include/yaml-cpp` 
`/usr/local/lib/cmake/yaml-cpp/` points to `../../Cellar/yaml-cpp/0.6.3_1/lib/cmake/yaml-cpp`

I'm fairly new to Cmake but It seems like this is where it would be looking.

dimaxano, 2020-07-07:
Did you tried solution from Troubleshooting page? https://openvslam.readthedocs.io/en/master/trouble_shooting.html
It helps quite often

wbrennan899, 2020-07-07:
@dimaxano Yes, but I get this warning whether is set it to ON / OFF

```
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_WITH_MARCH_NATIVE
```

RashidLadj, 2020-07-07:
I don't know if this resolve your problem , ut try, active OpenMP and recompil g2o with this script https://github.com/xdspacelab/openvslam/issues/239#issuecomment-648874135 #239